ros2/rcl #13: Error in stdatomic_helper.h while compiling with gcc version > 5

firesurfer, 2016-01-04:
I get the following error while compiling ROS2 with  gcc (Debian 5.3.1-4) 5.3.1 20151219

error: missing binary operator before token "("
 #elif defined(__clang__) && defined(__has_feature) && !__has_feature(c_atomic)

It compiles when i change line 23 in stdatomic_helper.h from

>#elif defined(__clang__) && defined(__has_feature) && !__has_feature(c_atomic)

to 

>#elif defined(__clang__) && defined(__has_feature) 

on my server with gcc (Debian 4.9.2-10) 4.9.2
it compiles without any errors.

wjwwood, 2016-01-05:
Interesting, thanks. I guess the evaluation of the conditional doesn't stop when `defined(__has_feature)` is false.

I suppose we can try something like this:

```c++
#elif defined(__clang__) && defined(__has_feature)
#if !__has_feature(c_atomic)
...
#endif
...
```

wjwwood, 2016-01-05:
@firesurfer if you have a chance, can you try the above and let me know if it works? I don't have a GCC 5 machine handy. If not I can set one up later.

firesurfer, 2016-01-05:
Hey,
if I change it to:
```c++
#elif defined(__clang__) && defined(__has_feature) 
#if !__has_feature(c_atomic)
// If Clang and no c_atomics (true for some older versions), use the compatability header.
#include "stdatomic_helper/gcc/stdatomic.h"
#endif
```
it compiles.

wjwwood, 2016-01-05:
I proposed this as a fix: https://github.com/ros2/rcl/pull/14

PR #14 (2016-01-05): change preprocess logic to fix error with gcc5
fixes #13
commit ec0e9306 (2016-01-05): change preprocess logic to fix error with gcc5

fixes #13
closed 2016-01-05 by wjwwood