ros2/rcl #15: Unused variable warning when compiling in `RelWithDebInfo`

gerkey, 2016-01-05:
The problem is that the variable is only referenced inside an `assert()`, which is presumably being compiled out under `RelWithDebInfo`.

~~~
Building C object CMakeFiles/rcl__rmw_opensplice_cpp.dir/src/rcl/timer.c.o
/home/gerkey/ros2_ws/src/ros2/rcl/rcl/src/rcl/wait.c: In function ‘__wait_set_clean_up’:
/home/gerkey/ros2_ws/src/ros2/rcl/rcl/src/rcl/wait.c:66:13: warning: variable ‘ret’ set but not used [-Wunused-but-set-variable]
   rcl_ret_t ret;
             ^
/home/gerkey/ros2_ws/src/ros2/rcl/rcl/src/rcl/wait.c: In function ‘rcl_wait’:
/home/gerkey/ros2_ws/src/ros2/rcl/rcl/src/rcl/wait.c:431:15: warning: variable ‘rcl_ret’ set but not used [-Wunused-but-set-variable]
     rcl_ret_t rcl_ret;
               ^
~~~

wjwwood, 2016-01-05:
I'd just add a `(void)ret;` after each of those lines.

wjwwood, 2016-01-05:
I'm looking into a pr for this.

dirk-thomas, 2016-02-05:
Can this be closed?

gerkey, 2016-02-17:
Confirmed fixed when building `RelWithDebInfo` on Linux.

closed 2016-02-17 by gerkey