ros2/rcl #22: rcl test failures when run with only opensplice on windows

tfoote, 2016-02-12:
We're seeing some test failures on the windows build when connext and fastrtps are disabled. 

They were first observed in #12 and #21 merge reviews and have been confirmed to be a bug in master: http://ci.ros2.org/job/ci_windows/1054/  

The failures are in these tests. 
TestNodeFixture.test_rcl_node_accessors
TestNodeFixture.test_rcl_node_life_cycle

In #21 @wjwwood has also observed it with connext too, still windows only.

jacquelinekay, 2016-02-12:
error output from one of the failing tests:

```
17: [ RUN      ] TestNodeFixture.test_rcl_node_accessors
17: ========================================================================================
17: Report      : ERROR
17: Date        : Fri Feb 12 10:56:10 Pacific Standard Time 2016
17: Description : Failed to connect to/create the domain with ID 42
17: Node        : leon
17: Process     : test_node__rmw_opensplice_cpp.exe <2724>
17: Thread      : 3004
17: Internals   : V6.4.140407OSS/3b83e82/3b83e82/gapi::gapi_domainParticipantFactory::gapi_domainParticipantFactory_create_participant/gapi_domainParticipantFactory.c/285/0/1455303
370.637426897
17: starting memory checking... not available
17: stopping memory checking... not available
17: C:\cygwin64\home\Jackie\dev\ros2\src\ros2\rcl\rcl\test\rcl\test_node.cpp(78): error: Value of: ret
17:   Actual: 100
17: Expected: 0
17: stopping memory checking... not available
17: [  FAILED  ] TestNodeFixture.test_rcl_node_accessors (5 ms)
17: [ RUN      ] TestNodeFixture.test_rcl_node_life_cycle
17: starting memory checking... not available
17: -- run_test.py: invocation failed: 'utf-8' codec can't decode byte 0xd2 in position 60: invalid continuation byte
17: stopping memory checking... not available
17: C:\cygwin64\home\Jackie\dev\ros2\src\ros2\rcl\rcl\test\rcl\test_node.cpp(237): error: Value of: ret
17:   Actual: 0
17: Expected: 101
17: Expected RCL_RET_NOT_INIT
17: stopping memory checking... not available
17: [  FAILED  ] TestNodeFixture.test_rcl_node_life_cycle (1314 ms)
```

tfoote, 2016-02-12:
@wjwwood and I looked into this some last night. 

This test code block is special cased for windows and opensplice: https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_node.cpp#L72-L82

We believe that the 2nd failure is a fragility of the test that the first failure is not cleaning up and deiniting the node when it asserts and is calling `rcl_node_fini` since it's scoped cleanup is registered after this assert. 

So when the next test tests that the node is not initied, it fails since the node was inited in the previous failed test. https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_node.cpp#L237

jacquelinekay, 2016-02-12:
ah, so it would be better to fini the node after the call to init that's expected to fail?

tfoote, 2016-02-12:
The problem is that it's hitting a gtest ASSERT incorrectly, and that's stopping the test, which prevents any following cleanup. If we fix the first one, it will fix the second one. 

We can't really register the scoped fini earlier since the init has not yet been known to be completed. We could check the return value and store it locally, and register the cleanup before asserting. But it's probably fine just to figure out why it's failing and fix it, then the test will pass again.

tfoote, 2016-02-17:
This is resolved with #24 and ros2/rmw#54

closed 2016-02-17 by tfoote