ros2/rcl #52: Linking errors for client_fixture and service_fixture tests

dhood, 2016-04-26:
The `service_fixture` and `client_fixture` tests are not linking properly on my machine. The rest work fine.

```
Linking CXX executable client_fixture__rmw_connext_cpp
cd /home/dhood/src/rcl/build/rcl/test && /usr/bin/cmake -E cmake_link_script CMakeFiles/client_fixture__rmw_connext_cpp.dir/link.txt --verbose=1
/usr/bin/c++    -std=c++11 -Wall -Wextra    CMakeFiles/client_fixture__rmw_connext_cpp.dir/rcl/client_fixture.cpp.o  -o client_fixture__rmw_connext_cpp  -L/home/dhood/rti_connext_dds-5.2.0/lib/x64Linux3gcc4.8.2 -rdynamic ../librcl__rmw_connext_cpp.so -ldl memory_tools/librcl_memory_tools.so /home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib/librmw_connext_shared_cpp.so /home/dhood/ros2_ws/install_isolated/rmw/lib/librmw.so /home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib/librosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/lib/librosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_cpp/lib/librmw_connext_cpp.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_opensplice_c.so /home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib/libbuiltin_interfaces__rosidl_typesupport_opensplice_cpp.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_opensplice_c.so /home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_opensplice_cpp.so -lnddsc -lnddscore -lnddscpp -lrticonnextmsgcpp -lcmxml -lcommonserv -ldcpsgapi -ldcpssac -lddsconfparser -lddsconf -lddsdatabase -lddsi2 -lddskernel -lddsosnet -lddsos -lddsserialization -lddsuser -lddsutil -ldurability -lspliced -ldcpssacpp -lpthread -ldl /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_opensplice_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_opensplice_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib/librmw_connext_shared_cpp.so /home/dhood/ros2_ws/install_isolated/rmw/lib/librmw.so /home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib/librosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/lib/librosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_cpp/lib/librmw_connext_cpp.so -ldl -lnddsc -lnddscore -lnddscpp -lrticonnextmsgcpp -lcmxml -lcommonserv -ldcpsgapi -ldcpssac -lddsconfparser -lddsconf -lddsdatabase -lddsi2 -lddskernel -lddsosnet -lddsos -lddsserialization -lddsuser -lddsutil -ldurability -lspliced -ldcpssacpp -lpthread -ldl -Wl,-rpath,/home/dhood/src/rcl/build/rcl:/home/dhood/src/rcl/build/rcl/test/memory_tools:/home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib:/home/dhood/ros2_ws/install_isolated/rmw/lib:/home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp/lib:/home/dhood/ros2_ws/install_isolated/rmw_connext_cpp/lib:/home/dhood/ros2_ws/install_isolated/builtin_interfaces/lib:/home/dhood/ros2_ws/install_isolated/example_interfaces/lib:/home/dhood/rti_connext_dds-5.2.0/lib/x64Linux3gcc4.8.2:/home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib 
/home/dhood/rti_connext_dds-5.2.0/lib/x64Linux3gcc4.8.2/libnddscore.so: undefined reference to `dlopen'
/home/dhood/rti_connext_dds-5.2.0/lib/x64Linux3gcc4.8.2/libnddscore.so: undefined reference to `dlclose'
/home/dhood/rti_connext_dds-5.2.0/lib/x64Linux3gcc4.8.2/libnddscore.so: undefined reference to `dlerror'
/home/dhood/rti_connext_dds-5.2.0/lib/x64Linux3gcc4.8.2/libnddscore.so: undefined reference to `dlsym'
```

`-ldl` is in the linking output, but I think it might be something similar to [this issue](http://stackoverflow.com/questions/20131138/cmake-add-ldl-at-end-of-link-stage-of-add-library)
I'm on Trusty with gcc version 4.8.4. I think the buildfarm has 4.6.3.

I have seen that the `memory_tools` tests are doing [something specific](https://github.com/ros2/rcl/blob/master/rcl/test/memory_tools/CMakeLists.txt#L16) with `dl` which might be relevant to these tests as well?

jacquelinekay, 2016-04-26:
https://github.com/ros2/rcl/blob/master/rcl/test/rcl_add_custom_executable.cmake#L41-Lundefined

It could be that the macro used the build these executables, `rcl_add_custom_executables`, does not link ` ${_ARG_APPEND_LIBRARY_DIRS}` to the executable, and the missing symbols are passed to the executable through memory tools, which is specified through that argument.

jacquelinekay, 2016-04-26:
I'm going to fix up various issues with the cmake infrastructure of rcl tests in an upcoming PR, I'll let you know when it's ready so you can test it locally.

dirk-thomas, 2016-04-26:
The build farm is using Trusty and therefore g++ 4.8. It should be exactly the same as you version locally.

Therefore I would guess that there must be a difference in:
* either what exactly code you compile (double check that everything is up-to-date and using the right branches)
* or how its build (try a clean build).

dhood, 2016-04-26:
I get this even after 
```
vcs pull
rm -r build_isolated install_isolated
src/ament/ament_tools/scripts/ament.py build --build-tests --isolated
```

Is this outdated? https://github.com/ros2/ros2/blob/ci_scripts/ros2_batch_job/linux_batch/__init__.py#L54 the jobs output that warning.

I did not install connext from debian packages. I'll try that.
Are they still available somewhere so I can update the broken links here https://github.com/ros2/ros2/wiki/Linux-Development-Setup#debian-packages-built-by-osrf (I grabbed them off the buildfarm)

dhood, 2016-04-26:
had also done `vcs custom --args checkout master` by the way

dirk-thomas, 2016-04-26:
The latest release from RTI was built with that compiler version and they embed that into the shell file (since they build many different combinations).

There is no public Debian package available since we are not allowed to distribute the package we built for internal use.

dhood, 2016-04-27:
update: I changed from the official connext packages to the debian ones and now my linking errors are gone. the debian packages might be using different version of connext

And I will remove that section from wiki entirely since it doesn't sound like it should be there

dirk-thomas, 2016-04-27:
The wiki section makes sense for us since it describes what to install if you have access. Therefore I would propose to keep it - eventually improve the note.

dhood, 2016-04-27:
ah ok then I'll just explain that it's only for OSRF (assuming that link is still right? I can't access that repo but might just not have the privileges)

closed 2016-04-28 by wjwwood