ros2/rcl #66: Linking errors for tests when using connext

dhood, 2016-06-21:
Am having linking issues for at least the following tests:
- rmw_connext_dynamic_cpp
  - test_client
  - service_fixture
  - client_fixture
  - test_allocator
  - test_time
  - test_publisher
- rmw_connext_cpp
  - test_services
  - service_fixture
  - client_fixture
  - test_client

no issues for opensplice

the error is:
```
[ 82%] Building CXX object test/CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/rcl/test_wait.cpp.o
cd /home/dhood/ros2_ws/build_isolated/rcl/test && /usr/bin/c++   -DRMW_IMPLEMENTATION=rmw_connext_dynamic_cpp -DRTI_LINUX -DRTI_UNIX -I/home/dhood/ros2_ws/src/ros2/rcl/rcl/include -I/home/dhood/ros2_ws/install_isolated/gtest_vendor/src/gtest_vendor/gtest-1.7.0/include -I/home/dhood/ros2_ws/install_isolated/rosidl_generator_c/include -I/home/dhood/ros2_ws/install_isolated/rmw/include -I/home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/include -I/home/dhood/rti_connext_dds-5.2.3/include -I/home/dhood/rti_connext_dds-5.2.3/include/ndds -I/home/dhood/ros2_ws/install_isolated/rosidl_generator_cpp/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/include/rosidl_typesupport_introspection_c/impl -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/include -I/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp/impl -I/home/dhood/ros2_ws/install_isolated/rcl_interfaces/include  -std=c++11 -Wall -Wextra   -o CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/rcl/test_wait.cpp.o -c /home/dhood/ros2_ws/src/ros2/rcl/rcl/test/rcl/test_wait.cpp
[ 83%] Linking CXX executable test_wait__rmw_connext_dynamic_cpp
cd /home/dhood/ros2_ws/build_isolated/rcl/test && /usr/bin/cmake -E cmake_link_script CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/link.txt --verbose=1
/usr/bin/c++    -std=c++11 -Wall -Wextra   CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/rcl/test_wait.cpp.o  -o test_wait__rmw_connext_dynamic_cpp  -L/home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2 ../gtest/libgtest.a ../gtest/libgtest_main.a ../librcl__rmw_connext_dynamic_cpp.so -ldl memory_tools/librcl_memory_tools.so /home/dhood/ros2_ws/install_isolated/rmw/lib/librmw.so /home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib/librmw_connext_shared_cpp.so /home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib/librosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/lib/librosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib/librosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_dynamic_cpp/lib/librmw_connext_dynamic_cpp.so -lnddsc -lnddscore -lnddscpp -lrticonnextmsgcpp -lpthread /usr/lib/x86_64-linux-gnu/libdl.so -lpthread /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_connext_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_connext_cpp.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_opensplice_c.so /home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib/librcl_interfaces__rosidl_typesupport_opensplice_cpp.so -lcmxml -lcommonserv -ldcpsgapi -ldcpssac -lddsconfparser -lddsconf -lddsdatabase -lddsi2 -lddskernel -lddsosnet -lddsos -lddsserialization -lddsuser -lddsutil -ldurability -lspliced -ldcpssacpp /home/dhood/ros2_ws/install_isolated/rmw/lib/librmw.so /home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib/librmw_connext_shared_cpp.so /home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib/librosidl_generator_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/lib/librosidl_typesupport_introspection_c.so /home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib/librosidl_typesupport_introspection_cpp.so /home/dhood/ros2_ws/install_isolated/rmw_connext_dynamic_cpp/lib/librmw_connext_dynamic_cpp.so -lnddsc -lnddscore -lnddscpp -lrticonnextmsgcpp -lpthread /usr/lib/x86_64-linux-gnu/libdl.so -ldl -Wl,-rpath,/home/dhood/ros2_ws/build_isolated/rcl:/home/dhood/ros2_ws/build_isolated/rcl/test/memory_tools:/home/dhood/ros2_ws/install_isolated/rmw/lib:/home/dhood/ros2_ws/install_isolated/rmw_connext_shared_cpp/lib:/home/dhood/ros2_ws/install_isolated/rosidl_generator_c/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_c/lib:/home/dhood/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib:/home/dhood/ros2_ws/install_isolated/rmw_connext_dynamic_cpp/lib:/home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2:/home/dhood/ros2_ws/install_isolated/rcl_interfaces/lib 
/home/dhood/ros2_ws/install_isolated/rmw_connext_dynamic_cpp/lib/librmw_connext_dynamic_cpp.so: undefined reference to `connext::RequesterParams::service_name(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/dhood/ros2_ws/install_isolated/rmw_connext_dynamic_cpp/lib/librmw_connext_dynamic_cpp.so: undefined reference to `connext::check_retcode(DDS_ReturnCode_t, char const*, RTILogMessage const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/build.make:149: recipe for target 'test/test_wait__rmw_connext_dynamic_cpp' failed
make[2]: *** [test/test_wait__rmw_connext_dynamic_cpp] Error 1
make[2]: Leaving directory '/home/dhood/ros2_ws/build_isolated/rcl'
CMakeFiles/Makefile2:1812: recipe for target 'test/CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/all' failed
make[1]: *** [test/CMakeFiles/test_wait__rmw_connext_dynamic_cpp.dir/all] Error 2
make[1]: Leaving directory '/home/dhood/ros2_ws/build_isolated/rcl'
Makefile:141: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/dhood/ros2_ws/build_isolated/rcl/cmake__build.sh && /usr/bin/make -j1' failed with exit code '2'

```

seems to be a problem with `rmw_connext_dynamic_cpp/lib/librmw_connext_dynamic_cpp.so`
same issue when using connext 5.2.0 instead of 5.2.3. 

looking into it

dhood, 2016-06-21:
similar error but for `example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_cpp.so`:

```
[ 70%] Linking CXX executable test_service__rmw_connext_cpp
/home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_cpp.so: undefined reference to `connext::RequesterParams::service_name(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/dhood/ros2_ws/install_isolated/example_interfaces/lib/libexample_interfaces__rosidl_typesupport_connext_cpp.so: undefined reference to `connext::check_retcode(DDS_ReturnCode_t, char const*, RTILogMessage const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_service__rmw_connext_cpp.dir/build.make:177: recipe for target 'test/test_service__rmw_connext_cpp' failed
make[2]: *** [test/test_service__rmw_connext_cpp] Error 1
CMakeFiles/Makefile2:1667: recipe for target 'test/CMakeFiles/test_service__rmw_connext_cpp.dir/all' failed
make[1]: *** [test/CMakeFiles/test_service__rmw_connext_cpp.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

dhood, 2016-06-22:
for reference:
```
dhood@osrf-esteve:~/ros2_ws [ros2_ws]$ nm -C /home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2/librticonnextmsgcpp.so | grep RequesterParams::service_name
0000000000015482 T connext::RequesterParams::service_name(std::string const&)
dhood@osrf-esteve:~/ros2_ws [ros2_ws]$ nm -C /home/dhood/rti_connext_dds-5.2.3/lib/x64Linux3gcc4.8.2/librticonnextmsgcpp.so | grep check_retcode
0000000000014b02 T connext::check_retcode(DDS_ReturnCode_t, char const*, RTILogMessage const&, std::string const&, bool)
```

dirk-thomas, 2016-06-22:
The problem has already been addressed:
https://github.com/ros2/rmw_connext/pull/175 You might want to make sure
that you have the latest version of "everything". If you have branches you
might want to rebase them.

Cheers,
- Dirk

dhood, 2016-06-22:
it's my first build on xenial; I'm using https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos, so should have everything up-to-date

I noticed that commit, but in my case it seems like it might be causing the problem? if I revert back to its parent the build runs fine for me. gcc 5.3.1

dhood, 2016-06-22:
I got connext 5.2.3 off the RTI website

dirk-thomas, 2016-06-22:
I have only tried the self-built Debian package as well as the evaluation version of Connext we have at OSRF. But the CMake code tests the library ABI and should choose the right one. I assume for the package from the website you should see this message: https://github.com/ros2/rmw_connext/pull/175/files#diff-f72dd578a73ca38cf3b97780449b385cR320

dhood, 2016-06-22:
hmm I don't see that message - I end up in the else statement (that I added) for some reason. I'll see if I can figure it out.

dhood, 2016-06-22:
I think I've figured out the cause - could I get your opinion on the appropriate fix @dirk-thomas 

The `try_compile` is failing, but not because of `#define _GLIBCXX_USE_CXX11_ABI 0`. 

Instead, it's failing for two reasons:

1. when [these variables](https://github.com/ros2/rmw_connext/blob/fa67e1b59a8bdaceca1d2dc9bcc1db840b65c45a/connext_cmake_module/cmake/Modules/FindConnext.cmake#L247) are put into the CMakeLists.txt generated from [check_abi.cmake](https://github.com/ros2/rmw_connext/blob/master/connext_cmake_module/cmake/check_abi.cmake#L19), they appear as `add_definitions(RTI_LINUX;RTI_UNIX)` instead of `add_definitions(-DRTI_LINUX;-DRTI_UNIX)`. just change them to have -D? (and the windows ones)

2. linking fails unless `-Wl,--no-as-needed` is added [here](https://github.com/ros2/rmw_connext/blob/master/connext_cmake_module/cmake/check_abi.cmake#L17), as in https://github.com/ros2/rcl/pull/55. could alternatively add a call to `dlopen` in `check_abi.cpp`.

then there's the issue of misinterpreting the result of the failed try_compile. Add another check which uses `_GLIBCXX_USE_CXX11_ABI 1` and ensure that at least one of them compiles?

dirk-thomas, 2016-06-22:
The current content of the variable `Connext_DEFINITIONS` is being used with `target_compile_definitions` which does not expect a leading `-D` before the preprocessor definitions. Therefore changing it to include the prefix is not a good idea imo.

Instead the try-compile-cmake file should use `target_compile_definitions` instead of `add_definitions`. Please try if the patch from ros2/rmw_connext#177 makes it work for you.

Adding `-Wl,--no-as-needed` to the try-compile-cmake sounds fine to me. It wasn't necessary for both of my use cases.

Doing both tests would certainly increase the robustness of the test. I am not sure if it is worth the extra effort once this try-compile works well.

dhood, 2016-06-22:
ok https://github.com/ros2/rmw_connext/pull/177 and https://github.com/ros2/rmw_connext/pull/178 together fix this, thanks

closed 2016-06-22 by dirk-thomas