ros2/rcl #81: Implement rcl_x_is_valid methods

firesurfer, 2016-09-23:
In order to track what was discussed in #78 I open this issue tracker. 

@wjwwood  suggested that in the future it might become necessary/useful to implement functions in the style `rcl_x_is_valid` .

An example would be:  `bool rcl_service_is_valid(rcl_service_t * service)` which would check if this service is valid. (Perhaps check if it's not null, check if it was initialized, and so on)

allenh1, 2016-12-23:
@tfoote Is this the kind of thing I could work on?

tfoote, 2016-12-23:
Yeah, We've tagged several issues ["help wanted"](https://waffle.io/ros2/ros2?search=help%20wanted) or also "unclaimed"](https://waffle.io/ros2/ros2?search=unclaimed) that you could look into.

There are many more, but we tried to pick out some good places for people to get involved.

allenh1, 2016-12-24:
We'll I'll start implementing this, then!

allenh1, 2016-12-26:
For now, I'm putting these in the `rcl.h` header -- is this appropriate?

allenh1, 2016-12-26:
Actually, after further investigation, I'm now thinking they should be implemented in several different header files (`bool rcl_service_is_valid(rcl_service_t * service)` should be in `service.h`, and similar).

Alternatively, I suppose I could put them in a `check.h` header, but I'll implement them in the above fashion in the meantime.

wjwwood, 2016-12-27:
Putting them in the same header with the type you're checking seems appropriate to me.

closed 2017-07-18 by allenh1