ros2/rcl #82: Reasons intra/interprocess message transport isn't implemented transparent in rcl

firesurfer, 2016-09-24:
After having a look at the implementation of the rclcpp I was wondering why there are special mechanisms implemented in the rclcpp for interprocess message transport and intraprocess message transport (there's an extra intra_process_manager)?  Wouldn't it be more convenient to have these mechanism transparently implemented in the rcl? By transparent I mean: The client lib on top of the rcl doesn't have to care how the message is transported.

wjwwood, 2016-09-24:
One reason is historical, we wanted to prototype intra-process communications before `rcl` was in place. At that time `rclcpp` called to `rmw` directly.

Another reason is that the intra-process communication makes heavy use of smart pointer semantics to help manage ownership of data. While it's not impossible to do this in C, it would be a bit more complicated.

Finally, if we move intra-process communications into `rcl` we need to sort out how to handle intra-process plus intra-language communications. Currently `rcl` doesn't know or care what types of objects are being passed through the `publish` and `take` calls, but if it starts holding those to deliver them to peers in the same process then it needs to know that the incoming data type is the right thing to return to the matching subscriber. For example, if you publish from a C++ program and store the C++ message for intra-process delivery but the subscriber is a C or Jave or Python program, then it wouldn't make sense to give it the C++ message, as a copy or the original. So then you need to convert from C++ to C or PyObject or whatever and need to know how to do that. If you have no conversion, then you need to first serialize the C++ object and then deserialize it in the type you want. All of this stuff about knowing when and how to convert between types is also associated with type masquerading.

wjwwood, 2016-09-24:
Also, the intra-process communications have to deal with multi-threading concerns, and up till now we've managed to avoid using threads or thread synchronization primitives in `rcl`, preferring to leave those things up to the client library implementations since each language has it's own mechanisms for dealing with that.

gbiggs, 2016-09-26:
It does sound like the use case for multi-language intra-process communication is too small to justify the complexity it would introduce.

clalancette, 2017-08-04:
@wjwwood @firesurfer Given this (old) discussion, should we close this ticket out?

firesurfer, 2017-08-04:
From my point of view it is still up to date because I still think it would be very useful for other language bindings, but regarding the extra work this would cause I think it's okey to close this ticket.

closed 2017-08-04 by clalancette