ros2/rcl #100: calling shutdown fails in examples_rclpy

dirk-thomas, 2016-12-19:
When running `examples_rclpy_minimal_service` and `examples_rclpy_minimal_client_async` with Connext the second one fails with the following error message after handling the response properly:

```
[rmw|error_handling.c:93] error string being overwritten: rcl node is invalid, rcl instance id does not match, at .../src/ros2/rcl/rcl/src/rcl/node.c:191
```

In that case the code tries to destroy the rcl client but the rcl node handle already has a mismatch with the underlying rmw node id since the Python code has called `shutdown` (https://github.com/ros2/examples/blob/b57f9591d96432671978f0a4be8a1f928bde5016/rclpy/services/minimal_client/client_async.py#L48).

clalancette, 2017-08-04:
Looking at the recent nightly builds, I believe this particular error has been fixed.  There are some other instances of "error string is being overwritten", but I am going to close this out and open new issues about those.

closed 2017-08-04 by clalancette

dirk-thomas, 2017-08-04:
> I am going to close this out and open new issues about those.

Please reference these new tickets here otherwise it is unclear where they are. Also consider spending time on resolving them during this sprint.

clalancette, 2017-08-04:
I'm not sure I agree; the new issue doesn't really have anything to do with this one.  But it's #155 , for reference.