ros2/rcl #119: replace rcl_impl_getenv with getenv from c_utilities

wjwwood, 2017-04-12:
See: https://github.com/ros2/rcl/blob/66208db876bba344a6733f4aa33451a028085612/rcl/src/rcl/common.c#L30

allenh1, 2017-06-22:
I'll get on this unless someone else wants it

mikaelarguedas, 2017-06-22:
We need to first change the implementation of getenv in rcutils to be thread safe or at least use thread local storage. So I think that we should discussed the refactor as a group rather than starting working on it right away. Ideally we will assign new tasks to everybody just after the release. In the meantime is it ok if you focus on the ones that are already assigned to you?

allenh1, 2017-06-22:
Yup -- sorry. Still getting used to how this works.

mikaelarguedas, 2017-06-22:
No worries, actually that's my bad for not updating this issue when I realized that the underlying implementation needed to be modified as well

mikaelarguedas, 2017-06-22:
need to discuss how we want to modify the implementation of `rcutils_get_env` before performing the refactor in rcl and in [rmw](https://github.com/ros2/rcl/pull/97) (see https://github.com/ros2/rcl/pull/125#discussion_r119451378)

mikaelarguedas, 2017-08-04:
status and implementation strategies are reported on the parent issue https://github.com/ros2/rcutils/issues/30

clalancette, 2019-10-21:
This was done in #502, so closing.

closed 2019-10-21 by clalancette