ros2/rcl #121: signatures in some time functions do not match rest of rcl

wjwwood, 2017-04-13:
For example, `rcl_init_time_source`:

https://github.com/ros2/rcl/blob/93ad1ae4fd05f9952db706f645e610ee377ebb37/rcl/include/rcl/time.h#L118

Should probably be `rcl_time_source_init`, as it is for `rcl_node_init` and `rcl_publisher_init` and so on.

wjwwood, 2017-04-13:
Also, many of the time functions call out to allocator functions directly like `calloc`:

https://github.com/ros2/rcl/blob/93ad1ae4fd05f9952db706f645e610ee377ebb37/rcl/src/rcl/time.c#L135

They should instead get passed an allocator (perhaps optionally) and use that to allocate memory.

mikaelarguedas, 2017-05-01:
Should this be put back in ready ? or closed in favor of a new issue with the remaining action item?

wjwwood, 2017-05-03:
Closed and rest of work put into https://github.com/ros2/rcl/issues/134.

closed 2017-05-03 by wjwwood