ros2/rcl #134: use allocator in rcl time related functions

wjwwood, 2017-05-03:
Some of the time functions call out to allocator functions directly like calloc:

https://github.com/ros2/rcl/blob/93ad1ae4fd05f9952db706f645e610ee377ebb37/rcl/src/rcl/time.c#L135

They should instead get passed an allocator (perhaps optionally) and use that to allocate memory.

Possibly the allocator structure should be extended to have calloc, so that the OS calloc can be used rather than emulating it with malloc and memset.

clalancette, 2017-08-04:
This is just me trying to understand, but what is the main purpose of having a custom allocator (here and elsewhere in the ROS2 codebase)?

Karsten1987, 2017-08-04:
the basic idea of it is to have a custom allocator strategy other than the inbuilt malloc and free. TLSF is one of such an allocator which has "better" strategies for allocating memory in a predefined pool of memory and thus is more suitable for real time critical apps. You'll often see that as well for STL containers: http://en.cppreference.com/w/cpp/container/vector

does that help?

mikaelarguedas, 2017-08-04:
The main reason is that according to the use case (very often realtime or memory constrained environment) a user may not want to use malloc but another strategy to allocate memory.

More information can be found on the design documents:
http://design.ros2.org/articles/realtime_background.html
http://design.ros2.org/articles/realtime_proposal.html

clalancette, 2017-08-04:
Makes sense.  Typically I think about using the LD_PRELOAD trick to replace malloc/free, but this plumbing is a more elegant solution.  Thanks.

wjwwood, 2017-08-07:
@clalancette That's right, we've used the `LD_PRELOAD` trick before to try things like counting the malloc calls or to use tlsf, but this approach (while more verbose) has the advantage of portability (doing these tricks differs on macOS as well as Windows) and being easier to test things. Since the allocator is injected into the function we can easily setup a failing allocator for just one part of the system, making it easier to unit test.

So I'm not sure it's worth it in the end (it's certainly uncommon), but hopefully it will pay off.

clalancette, 2019-10-21:
I'm fairly certain this is already done, as there are no instances of open-coded `calloc`, `malloc`, or `realloc` in time.c (or anywhere else in rcl, for that matter).  So I'll close this out.

closed 2019-10-21 by clalancette