ros2/rcl #162: rcl_init called multiple times

dirk-thomas, 2017-08-31:
_From @BrannonKing on March 30, 2017 15:50_

I'm getting the error below when attempting to use rmw_opensplice in the current (Mar 29) codebase. I'm running a Gazebo simulation. I'm guessing that Gazebo loads all the plugins into the same process, which causes this error. It seems I can eat the error and still run fine. I just wondered if we should track the initialization state in the RMW.

```
[rmw|error_handling.c:93] error string being overwritten: rcl_init called while already initialized, at src/ros2/rcl/rcl/src/rcl/rcl.c:69
failed to initialize rmw implementation: rcl_init called while already initialized, at src/ros2/rcl/rcl/src/rcl/rcl.c:69
```

_Copied from original issue: ros2/rmw_opensplice#165_

dirk-thomas, 2017-08-31:
_From @BrannonKing on August 31, 2017 20:16_

I suppose catching the exception in Gazebo is fine. I just wanted the logging removed (since it already throws an exception). BTW, this should move to rcl; it's logged in the wrong spot.

dirk-thomas, 2017-08-31:
@BrannonKing Can you please provide more information about how the API is being used and if possible how to reproduce the problem.

BrannonKing, 2017-08-31:
I start up multiple Gazebo plugins. Each one has to call rclcpp::init because you don't know which plugin will start first. They all run in the same process, but init prints a message and throws an error about having been called already for every subsequent plugin. I can eat the exception but can't hide the message.

wjwwood, 2017-08-31:
You could just ignore the ones that fail to init for this particular reason (assuming that some other code managed to init first). However we should make sure `rclcpp::init` is thread-safe. `rcl_init` is not:

https://github.com/ros2/rcl/blob/master/rcl/include/rcl/rcl.h#L108

dirk-thomas, 2017-08-31:
I think that is what he is trying to do but the calls which throw also print an error message which he currently can't avoid. I will come up with a PR to fix this...

wjwwood, 2017-08-31:
The rcl error just needs to be cleared here: https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L168

@dirk-thomas I'll let you do that since you already said you were working on it.

But we should also make sure it is thread-safe, because right now it is not I think.

dirk-thomas, 2017-08-31:
> The rcl error just needs to be cleared here: https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L168

Indeed: ros2/rclcpp#366

closed 2017-08-31 by dirk-thomas

wjwwood, 2017-08-31:
I'll just reiterate that `rclcpp::init()` is not thread-safe. So we either could use a pr to make it so, or @BrannonKing you should take care to protect it in your code.

BrannonKing, 2017-09-01:
I'm quite certain that Gazebo does not load its plugins in parallel. I don't think that multithreaded access will be an issue there.