ros2/rcl #204: rcl_difference_times() function 32 bit overflow

kenny-y, 2017-12-28:
`rcl_difference_times()` function can't process time diff > 2^31

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Ardent Apalone
- Version or commit hash:
  - NA
- DDS implementation:
  - Default
- Client library (if applicable):
  - rcl

#### Steps to reproduce issue
This test case would pass
```
Start: sec = 1514423496, nanosec = 0
Finish: sec = 1514423498, nanosec = 147483647
64 bit .nanoseconds: (1514423496000000000 - 1514423498147483647)
rcl_difference_times: delta.nanoseconds = 2147483647
Human readable result: { sec: 2, nanosec: 147483647 }
```

This test case would fail
```
Start: sec = 1514423496, nanosec = 0
Finish: sec = 1514423498, nanosec = 147483648
64 bit .nanoseconds: (1514423496000000000 - 1514423498147483648)
rcl_difference_times: delta.nanoseconds = -2147483648  // This is the issue
Human readable result: { sec: -2, nanosec: -147483648 }
```

#### Expected behavior
that `rcl_difference_times()` function can process diff > 2 ^ 31 (e.g. 3 seconds or greater)

#### Actual behavior
32 bit signed integer overflow

#### Additional information
I found this when I'm writing [Node.js bindings to rcl](https://github.com/RobotWebTools/rclnodejs)

wjwwood, 2017-12-28:
Thanks for pointing it out @kenny-y. A pr with a test for this and a fix would be greatly appreciated. Otherwise we'll get to it as soon as we have time.

kenny-y, 2017-12-29:
I'll see what I can do after the new year vacation.

closed 2018-03-26 by mjcarroll