ros2/rcl #247: [YAML parser] Support for multiple parameter files

anup-pem, 2018-06-01:
#235 takes of supporting parsing single YAML parameter file.

The above work needs to be extended to support multiple parameter files. This helps where a vendor can ship their product with a default parameter file and customer can use their parameter file if they want to change some of the default values. The files will be parsed one after the other updating the same C parameter structure. If the same node/parameter exists in the structure, the newer values will update the older values. If it is a new node/parameter, it needs to get appended. Also, if the parameter struct is already present which was created by previous rcl_parse_yaml_file(), the later calls updates the structure

```
/// \brief Parse the YAML file, initialize and populate params_st
/// \param[in] file_patha is the list of paths to the YAML files
/// \param[in/out] params_st points to the populated paramter struct
/// \return true on success and false on failure
RCL_YAML_PARAM_PARSER_PUBLIC
bool rcl_parse_yaml_file(
  const char ** file_paths,
  rcl_params_t * params_st);
```

KeisukeShima, 2021-11-01:
@ivanpauno  Thanks for the reply. I was a little mistaken.
I can pass multiple parameter files, and if the same key exists, it will be overwritten by the later one.

What I was testing was an array of strings. I expected the elements of the array to be merged, but in fact they were overwritten by the values in the later file, which confused me. But this is the correct behavior. 
Forget my previous comment and close this ticket. :pray:

ivanpauno, 2021-11-01:
No problem, closing the ticket then!

closed 2021-11-01 by ivanpauno