ros2/rcl #262: __ns remapping not working when specified for a particular node

dhood, 2018-06-19:
This works:
```
]$ ros2 run demo_nodes_cpp talker talker:__node:=new_node_name
[INFO] [new_node_name]: Publishing: 'Hello World: 1'
^Csignal_handler(2)
```

As does this:
```
$ ros2 run demo_nodes_cpp talker __ns:=/new_ns
[INFO] [new_ns.talker]: Publishing: 'Hello World: 1'
[INFO] [new_ns.talker]: Publishing: 'Hello World: 2'
```

But this doesn't seem to:
```
$ ros2 run demo_nodes_cpp talker talker:__ns:=/new_ns
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [talker]: Publishing: 'Hello World: 2'
^Csignal_handler(2)
```

There's a chance this combination isn't expected to work, but I think it might be.

wjwwood, 2018-06-19:
I would also consider that to be a bug.

sloretz, 2018-06-19:
definitely a bug

There's a unit test for `rcl_remap_node_namespace()` https://github.com/ros2/rcl/blob/c9fe312674f8c5a2c5b40b6208c830f4ca8c40ff/rcl/test/rcl/test_remap.cpp#L55-L86

but there's not an integration test for it https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_remap_integration.cpp

dhood, 2018-06-19:
Yeah I have one prepared that fails, will open a pr for the test