ros2/rcl #296: Name remapping specific to a fully-qualified node's name

dhood, 2018-09-11:
Remapping can be passed to a specific node with e.g. `talker:chatter:=my_chatter`.
Suppose I have multiple nodes named `chatter` in a process, but in different namespaces.
Is the ability to address the nodes individually implemented yet, so that I can do e.g. `/ns1/talker:chatter:=my_chatter`? I think not; am looking for confirmation. @sloretz

sloretz, 2018-09-11:
The node-name prefix handling assumes node names are unique within a process before remapping.

dhood, 2018-09-11:
Usually when we refer to node name uniqueness we are actually referring to "ns+node name" uniqueness. To clarify your statement: the remapping process is assuming there is only one `talker` in the process, in _any_ namespace, and it is not possible to address them individually, is that right?

sloretz, 2018-09-11:
> To clarify your statement: the remapping process is assuming there is only one talker in the process, in any namespace, and it is not possible to address them individually, is that right?

Correct

wjwwood, 2018-09-11:
I think it has to consider the node namespace as well, because having two nodes with the same name but in different namespaces will be (has been in the past) a common case.

mikaelarguedas, 2018-09-11:
> Usually when we refer to node name uniqueness we are actually referring to "ns+node name" uniqueness. 

:+1:

> I think it has to consider the node namespace as well, because having two nodes with the same name but in different namespaces will be (has been in the past) a common case.

:+1:

Agreed I think it should consider the namespace and assume uniqueness of "ns+node name".

I often use the example of a stereo image processing process that is a basic/common example of a process having 2 instances of a "camera" node under different namespaces (`camera_left`, `camera_right`).

iuhilnehc-ynos, 2020-09-16:
It seems that this feature is still not supported in ros2. I'd like to contribute to this issue.

```
$ ros2 run demo_nodes_cpp talker --ros-args --remap talker:chatter:=my_chatter
[INFO] [1600251961.967244593] [talker]: Publishing: 'Hello World: 1'
[INFO] [1600251962.967163181] [talker]: Publishing: 'Hello World: 2'

$ ros2 topic list
/my_chatter

but failed to run the following command line,

$ ros2 run demo_nodes_cpp talker --ros-args --remap /talker:chatter:=my_chatter
[ERROR] [1600251921.506736587] [rcl]: Failed to parse global arguments

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Couldn't parse remap rule: '--remap /talker:chatter:=my_chatter'. Error: Expected lexeme type 19, got 5 at index 7, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:238, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/arguments.c:371'

with this new error message:

  'context is zero-initialized, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/context.c:52'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1600251921.506782928] [rclcpp]: failed to finalize context: context is zero-initialized, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/context.c:52
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: error not set

```

iuhilnehc-ynos, 2020-09-21:
Actually, I found `--param` can also support this feature. After this issue, I'll create a new PR for `--param`.

iuhilnehc-ynos, 2020-09-22:
To remap __node and __ns with `a fully-qualified node's name`, such as `-r /my_talker:__ns:=/ns1 -r /talker:__node:=my_talker` (current remapping order) or `-r /talker:__ns:=/ns1 -r /ns1/talker:__node:=my_talker`(if remmaping namespace before node name, https://github.com/ros2/rcl/issues/806), I need to update two functions by adding `namespace`.
https://github.com/ros2/rcl/blob/8bcada7b884693644d7ba2e3bd2c22fd9541eba7/rcl/include/rcl/remap.h#L199-L207
https://github.com/ros2/rcl/blob/8bcada7b884693644d7ba2e3bd2c22fd9541eba7/rcl/include/rcl/remap.h#L240-L248

iuhilnehc-ynos, 2020-09-22:
@wjwwood

I have created a new PR for this issue at https://github.com/ros2/design/pull/299 that also including https://github.com/ros2/rcl/issues/806, could you please help review it?