ros2/rcl #302: [YAML Parser] Empty strings not allowed

Myzhar, 2018-09-26:
- Operating System:
  -  Ubuntu 16.04...)
- Installation type:
  -  from source
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp, rclpy

#### Steps to reproduce issue

In YAML file:
``` svo_file: "" ```
or
``` svo_file: '' ``` 
or
``` svo_file:  ``` 

In the C++ code:
```
        rclcpp::Parameter paramVal;
        std::string paramName;

        paramName = "general.svo_file";
        if (get_parameter(paramName, paramVal)) {
            mSvoFilepath = paramVal.as_string();
        } else {
            RCLCPP_WARN(get_logger(), "The parameter '%s' is not available, using the default value", paramName.c_str());
        }
        RCLCPP_INFO(get_logger(), " * SVO: `%s`", mSvoFilepath.c_str());
```

#### Expected behavior
Read parameter as an empty std::string
#### Actual behavior
Runtime error:
```what():  Failed to parse parameters from file '/home/walter/devel/ros2_ws/install/stereolabs_zed/share/stereolabs_zed/config/common.yaml': No value at line 12, at /home/walter/devel/ros2_ws_src/src/ros2/rcl/rcl_yaml_param_parser/src/parser.c:840```

#### Additional information
In my particular case the parameter is a filename. If it is filled I load the file, otherwise I open an hardware device. But I cannot set an empty parameter.

clalancette, 2018-10-11:
@Myzhar Just FYI, after discussing this, we think that:

```
svo_file: ""
```
and
```
svo_file: ''
```
Should work, while:
```
svo_file:
```

Should not (it's either a syntax error or a null entry, it's not entirely clear to me which one).

Myzhar, 2018-10-11:
I tried both as reported in the issue and both give the same error "No value at line XX"

avrabe, 2019-05-24:
Had the same problem. #450 is my fix for it.

PR #450 (2019-05-24): Allow empty strings if they are quoted.
Fixes: #302
closed 2019-05-24 by dirk-thomas