ros2/rcl #309: Error when requesting transitions of lifecycle

fmrico, 2018-10-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04 
- Installation type:
  - binaries and source  
- Version or commit hash:
  - bouncy and 451bf4a1a41ad87920006a974a4d3b8ca528847b
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
In terminal 1: 
$ ros2 run lifecycle lifecycle_talker

In terminal 2: 
$ ros2 run lifecycle lifecycle_service_client_py.py change_state --change-state-args configure lc_talker
$ ros2 run lifecycle lifecycle_service_client_py.py change_state --change-state-args activate lc_talker
$ ros2 run lifecycle lifecycle_service_client_py.py change_state --change-state-args shutdown lc_talker
```

#### Expected behavior
Transition from active to shuttingdown 
#### Actual behavior

```
[rcl_lifecycle]: No transition matching 5 found for current state unconfigured
[ERROR] []: Unable to start transition 5 from current state unconfigured: Transition is not registered., at /home/paco/ros2/ros2_ws/src/ros2/rcl/rcl_lifecycle/src/rcl_lifecycle.c:332

```
#### Additional information

These trensition messages are defined in rcl_interfaces/lifecycle_msgs/msg/Transition.msg:14 
```
uint8 TRANSITION_SHUTDOWN = 5
uint8 TRANSITION_DESTROY = 6
```

Actually, demos/lifecycle/src/lifecycle_service_client_py.py:40 uses it
```
    elif change_state_args == 'shutdown':
        req.transition.id = Transition.TRANSITION_SHUTDOWN
```

but these transitions are never registered. I have debugged this by adding these lines just at the end of function rcl_lifecycle_register_transition(..) at rcl/rcl_lifecycle/src/transition_map.c:154: 

```
  const char *lstart = rcl_lifecycle_get_state(transition_map, transition.start->id)->label;
  const char *lgoal = rcl_lifecycle_get_state(transition_map, transition.goal->id)->label;
  const char *ltrans = rcl_lifecycle_get_transitions(transition_map, transition.id)->label;

  RCUTILS_LOG_INFO_NAMED(ROS_PACKAGE_NAME,
    "registering transition %u (%s) state %u (%s) -> %u (%s)\n",
      transition.id, ltrans,
      transition.start->id, lstart,
      transition.goal->id, lgoal);

  return RCL_RET_OK;
}
```

and the output, when I start the node, is: 

```
[INFO] [rcl_lifecycle]: registering transition 1 (configure) state 1 (unconfigured) -> 10 (configuring)
[INFO] [rcl_lifecycle]: registering transition 10 (transition_success) state 10 (configuring) -> 2 (inactive)
[INFO] [rcl_lifecycle]: registering transition 11 (transition_failure) state 10 (configuring) -> 1 (unconfigured)
[INFO] [rcl_lifecycle]: registering transition 12 (transition_error) state 10 (configuring) -> 15 (errorprocessing)
[INFO] [rcl_lifecycle]: registering transition 2 (cleanup) state 2 (inactive) -> 11 (cleaningup)
[INFO] [rcl_lifecycle]: registering transition 20 (transition_success) state 11 (cleaningup) -> 1 (unconfigured)
[INFO] [rcl_lifecycle]: registering transition 21 (transition_failure) state 11 (cleaningup) -> 2 (inactive)
[INFO] [rcl_lifecycle]: registering transition 22 (transition_error) state 11 (cleaningup) -> 15 (errorprocessing)
[INFO] [rcl_lifecycle]: registering transition 3 (activate) state 2 (inactive) -> 13 (activating)
[INFO] [rcl_lifecycle]: registering transition 30 (transition_success) state 13 (activating) -> 3 (active)
[INFO] [rcl_lifecycle]: registering transition 31 (transition_failure) state 13 (activating) -> 2 (inactive)
[INFO] [rcl_lifecycle]: registering transition 32 (transition_error) state 13 (activating) -> 15 (errorprocessing)
[INFO] [rcl_lifecycle]: registering transition 4 (deactivate) state 3 (active) -> 14 (deactivating)
[INFO] [rcl_lifecycle]: registering transition 40 (transition_success) state 14 (deactivating) -> 2 (inactive)
[INFO] [rcl_lifecycle]: registering transition 41 (transition_failure) state 14 (deactivating) -> 3 (active)
[INFO] [rcl_lifecycle]: registering transition 42 (transition_error) state 14 (deactivating) -> 15 (errorprocessing)
[INFO] [rcl_lifecycle]: registering transition 50 (shutdown) state 1 (unconfigured) -> 12 (shuttingdown)
[INFO] [rcl_lifecycle]: registering transition 51 (shutdown) state 2 (inactive) -> 12 (shuttingdown)
[INFO] [rcl_lifecycle]: registering transition 52 (shutdown) state 3 (active) -> 12 (shuttingdown)
[INFO] [rcl_lifecycle]: registering transition 53 (transition_success) state 12 (shuttingdown) -> 4 (finalized)
[INFO] [rcl_lifecycle]: registering transition 54 (transition_failure) state 12 (shuttingdown) -> 4 (finalized)
[INFO] [rcl_lifecycle]: registering transition 55 (transition_error) state 12 (shuttingdown) -> 15 (errorprocessing)
[INFO] [rcl_lifecycle]: registering transition 60 (transition_success) state 15 (errorprocessing) -> 1 (unconfigured)
[INFO] [rcl_lifecycle]: registering transition 61 (transition_failure) state 15 (errorprocessing) -> 4 (finalized)
[INFO] [rcl_lifecycle]: registering transition 62 (transition_error) state 15 (errorprocessing) -> 4 (finalized)
```

No transition 5 is registered

fmrico, 2018-10-27:
I have provided a fix in #310 if this is finally considered a bug.

Best

Karsten1987, 2018-10-27:
@fmrico thank you for reporting this. That's my bad. The python script you're using should actually be deprecated and deleted. It is still using the old API and transitions.

Can you try the following (recommended) way of changing the states:

```
ros2 lifecycle set /lc_talker configure
ros2 lifecycle set /lc_talker activate
ros2 lifecycle set /lc_talker shutdown
```

that works for me.

fmrico, 2018-10-27:
Thank you @Karsten1987 . I will close de PR.

clalancette, 2018-10-29:
I'm going to change the title of this bug to more closely reflect the fact that we want to remove `lifecycle_service_client_py.py`

closed 2018-11-01 by Karsten1987