ros2/rcl #347: Tests are failing when security is enabled

thomas-moulard, 2018-12-01:
Many tests fail because ROS_SECURITY_ROOT_DIRECTORY is not set when the tests are running.

```
The following tests FAILED:
          2 - cppcheck (Failed)
          6 - xmllint (Failed)
          7 - test_client__rmw_fastrtps_cpp (Failed)
         10 - test_get_node_names__rmw_fastrtps_cpp (Failed)
         13 - test_graph__rmw_fastrtps_cpp (Failed)
         14 - test_count_matched__rmw_fastrtps_cpp (Failed)
         16 - test_node__rmw_fastrtps_cpp (Failed)
         19 - test_remap_integration__rmw_fastrtps_cpp (Failed)
         21 - test_publisher__rmw_fastrtps_cpp (Failed)
         22 - test_service__rmw_fastrtps_cpp (Failed)
         23 - test_subscription__rmw_fastrtps_cpp (Failed)
         25 - test_namespace__rmw_fastrtps_cpp (Failed)
         26 - test_services__rmw_fastrtps_cpp (Failed)
         28 - test_client__rmw_opensplice_cpp (Failed)
         31 - test_get_node_names__rmw_opensplice_cpp (Failed)
         34 - test_graph__rmw_opensplice_cpp (Failed)
         35 - test_count_matched__rmw_opensplice_cpp (Failed)
         37 - test_node__rmw_opensplice_cpp (Failed)
         40 - test_remap_integration__rmw_opensplice_cpp (Failed)
         42 - test_publisher__rmw_opensplice_cpp (Failed)
         43 - test_service__rmw_opensplice_cpp (Failed)
         44 - test_subscription__rmw_opensplice_cpp (Failed)
         46 - test_namespace__rmw_opensplice_cpp (Failed)
         47 - test_services__rmw_opensplice_cpp (Failed)
         51 - test_timer__rmw_opensplice_cpp (Failed)
```

dirk-thomas, 2018-12-01:
The tests are not meant to be run with security enabled. Also I doubt that the `cppcheck` and `xmllint` failures are related to security.

thomas-moulard, 2018-12-01:
`cppcheck` and `xmllint` being put aside, what is your recommendation to ensure that we don't introduce regressions while working on sros?

dirk-thomas, 2018-12-01:
I would think that it is unrealistic that existing tests pass when any security related restrictions are in place. So I think you will need security specific tests for that - e.g. like these: https://github.com/ros2/system_tests/tree/master/test_security

thomas-moulard, 2018-12-01:
Thanks for the pointer, I'll take a look at that!

closed 2018-12-01 by thomas-moulard