ros2/rcl #363: bus error on armv7l

abilbaotm, 2018-12-19:
## Bug report

Hi, I have compiled natively ros2 crystal on a raspberry pi 3b and seems that there is a bug somewhere, when I try to launch a demo talker I get a bus error, seems something related with some `Alignment trap`. I open this issue in `rcl` repo because the exception is raised when `rcl_get_zero_initialized_context` is called.

**Required Info:**

- Operating System:
  - 2018-11-13-raspbian-stretch-full
- Installation type:
  - from sources, native build
- Version or commit hash:
  - 657d9e84c73e4268176efd163e96fda73c1a76d9
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rcl

#### Steps to reproduce issue

```
colcon build --symlink-install --packages-ignore qt_gui_cpp rqt_gui_cpp --packages-up-to demo_nodes_cpp --executor sequential --cmake-args " -DBUILD_TESTING=OFF" --event-handlers console_direct+
# source ros2 install
./build/demo_nodes_cpp/talker
```

#### Expected behavior
demo_nodes_cpp talker should be launch

#### Actual behavior
An exception is raised: "Bus error"
gdb output:
```
gdb ./build/demo_nodes_cpp/talker
r
Program received signal SIGBUS, Bus error.
0x76edb80c in rcl_get_zero_initialized_context () from /home/pi/ros2_ws/install/rcl/lib/librcl.so
```
journalctl will output the next:
```
dic 19 17:57:53 raspberrypi kernel: Alignment trap: not handling instruction e1b20f9f at [<ffff0f70>]
dic 19 17:57:53 raspberrypi kernel: Unhandled fault: alignment exception (0x001) at 0x7691f83c
dic 19 17:57:53 raspberrypi kernel: pgd = b9314000
```

ahcorde, 2018-12-20:
I have commented out this line https://github.com/ros2/rcl/blob/master/rcl/src/rcl/context.c#L41

```
  atomic_init((atomic_uint_least64_t *)(&context.instance_id_storage), 0);
```

and it's working

ArkadiuszNiemiec, 2019-01-03:
I can confirm that the problem exists. Tested on two separate RPi 3 with Raspbian.
Also the fix mentioned by @ahcorde worked for me as well.

closed 2019-01-03 by sloretz

Vigasto, 2019-01-16:
Fix https://github.com/ros2/rcl/issues/363#issuecomment-448924124 also works on Odroid XU4 running Armbian

PR #365 (2019-01-03): Ensure that context instance id storage is aligned correctly
On 32 bit ARMv7l, casting of the context instance id to an `atomic_uint_least64_t` and initialising it results in SIGBUS if it is not 8 byte aligned. This change fixes that by ensuring that `instance_id_storage` is aligned as `uint_least64_t`.

Closes #363