ros2/rcl #374: rcl_init segfaults on non ASCII command line arguments if LANG is not set

hidmic, 2019-01-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 Bionic Beaver
- Installation type:
  - Binaries and from source
- Version or commit hash:
  - Crystal and current [ros2.repos](https://github.com/ros2/ros2/commit/149b0e50083d25d1002e7dfc01af95ddb2c5b14d)

#### Steps to reproduce issue

1. Unset the `LANG` environment variable if set.
  ```sh
  unset LANG
  ```
2. Start some node.
  ```sh
  ros2 run demo_nodes_cpp talker
  ```
3. Try to set a string parameter with non-ASCII value
  ```sh
  ros2 param set /talker value 'Ragnarök'
  ```
#### Expected behavior
CLI should throw `UnicodeEncodeError` upon message conversion and serialization.

#### Actual behavior
CLI segfaults within `rcl_init`:

```
Program received signal SIGSEGV, Segmentation fault.                                                        [9/601]
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:62
62      ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
(gdb) bt  
#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:62
#1  0x00007ffff5ee3daa in rcl_init (argc=2, argv=0xab36d0, options=0x7fffffff2a00, context=0x7ffff7f5ddc8)
    at /home/michel/Workspaces/ros2_base_ws/src/ros2/rcl/rcl/src/rcl/init.c:97
#2  0x00007ffff67b75e9 in rclpy_init (_unused_self=0x7ffff7e2e458, args=0x7ffff7e25dc8)
    at /home/michel/Workspaces/ros2_base_ws/src/ros2/rclpy/rclpy/src/rclpy/_rclpy.c:529
...
#1  0x00007ffff5ee3daa in rcl_init (argc=2, argv=0xab36d0, options=0x7fffffff2a00, context=0x7ffff7f5ddc8)
    at /home/michel/Workspaces/ros2_base_ws/src/ros2/rcl/rcl/src/rcl/init.c:97
97            size_t argv_i_length = strlen(argv[i]);
```
#### Additional information
Above situation only arises when system locale settings are not properly configured as specified in the [installation instructions](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#system-requirements). From that perspective, this may not be a bug, and fixing it not a fix but an enhancement to better deal with such conditions.

dirk-thomas, 2019-02-07:
> CLI should throw `UnicodeEncodeError` upon message conversion and serialization.

The problem is not happening in message conversation / serialization. The problem already happens while passing the command line arguments.

hidmic, 2019-02-11:
Alright, small fixes are ready!

PR #388 (2019-02-12): Enforce non-null argv values on rcl_init().
Fixes #374.
closed 2019-02-12 by hidmic