ros2/rcl #431: Re-enable rcl unit test "test_events" for macOS

mm318, 2019-05-03:
The `test_events` unit test in `rcl` was flaky when run on the macOS build machine, so it was disabled (intended to be temporarily disabled). This issue to is track re-enabling the test after improving its robustness.

dirk-thomas, 2019-05-03:
For reference: https://github.com/ros2/rcl/pull/408/commits/209ec28cda85e4ad6477459eda6f60c67d861a4d

sloretz, 2019-05-03:
Quick check on OSX  testing just rcl @mm318 FYI

[image]

mm318, 2019-05-17:
The tests passed once (https://github.com/ros2/rcl/pull/433#issuecomment-493159962), got merged, and then failed again (https://github.com/ros2/build_cop/issues/199).

jacobperron, 2019-05-17:
I'm not sure if it's of any consequence, but the reported failure on build cop is a release build.

clalancette, 2019-05-17:
> I'm not sure if it's of any consequence, but the reported failure on build cop is part of a release build.

Yeah, that's almost certainly the case.  The release builds (particularly on macOS) expose a lot more bugs than the debug builds.

mm318, 2019-05-17:
Are there other unit tests that create a context, node, and publisher/subscriber pair, send a `Strings` message, and lastly tears everything down, and then repeat a couple of times for the remaining test cases?

Is there some way to run this test with `--gtest_shuffle`? I've noticed that it's always the last two test cases that fail, with the subscriber failing to receive a published message.

I feel like this test failure is exposing a bigger problem (on macOS) than the test itself.

jacobperron, 2019-05-17:
[test_communication/test/test_publisher_subscriber](https://github.com/ros2/system_tests/blob/master/test_communication/test/test_publisher_subscriber.py.in) launches pub/sub pairs for `rclcpp` and `rclpy` with all test messages. But I'm not aware of analogous  communication tests strictly in C.

> Is there some way to run this test with --gtest_shuffle? I've noticed that it's always the last two test cases that fail, with the subscriber failing to receive a published message

Try setting the env variable GTEST_SHUFFLE=1 ([reference](https://github.com/google/googletest/blob/master/googletest/docs/advanced.md#shuffling-the-tests)).

nburek, 2019-05-21:
Are https://github.com/ros2/build_cop/issues/199 and this tracking the same issue? Should one be closed in preference of the other?

mm318, 2019-05-21:
Yeah, let's close this one, as this test is already enabled, which is why it showed up under `build_cop`.

closed 2019-05-21 by mm318