ros2/rcl #449: Remap node name to fully qualified name

jacobperron, 2019-05-22:
## Feature request

#### Feature description

When using a fully qualified name for remapping, the node name is not remapped. For example:

```
ros2 run demo_nodes_cpp talker __node:=/foo
```

The above command does not change the node name, but I would expect the output of `ros2 node list` to contain `/foo`.

I'm not sure if this is intended to work as I expect, but it seems reasonable to me. 

If this is **not** intended to work, then I think there should be a warning logged indicating that the remap failed. Similar to the warning when failing to remap the namespace:

```
$ ros2 run demo_nodes_cpp talker __ns:=foo
[WARN] [rcl]: Namespace not remapped to a fully qualified name (found: foo)
```

sloretz, 2019-05-22:
Remapping to a fully qualified name requires `__ns:=/` and `__node:=foo`. 

I would have sworn there was already a ticket where it was talked about making `__node:=` accept a fully qualified name, but I can't find it. IIRC the trouble with `__node` taking a fully qualified name is what to do when a different namespace is given with `__ns:=`? Which namespace do topics/services use if given as relative names or `~/`?

See also http://design.ros2.org/articles/static_remapping.html#supporting-change-the-node-name

jacobperron, 2019-05-23:
I think it's okay to limit the name remap to to not accept fully qualified names, but it would be nice to log a warning for invalid names.

PR #454 (2019-06-05): Log warning when remapping to an invalid node name
Similar to the warnings when remapping to invalid namespaces, this better communicates failures to the user.

Resolves #449
commit 79b3ec10 (2019-06-05): Log warning when remapping to an invalid node name (#454)

Similar to the warnings when remapping to invalid namespaces, this better communicates failures to the user.

Resolves #449
closed 2019-06-05 by jacobperron