ros2/rcl #460: Support passing parameters as command line arguments

hidmic, 2019-06-13:
## Feature request

#### Feature description

Currently, the only way to pass node parameters over the command line is via a YAML file. To improve usability, we'd like to be be able to pass key value pairs directly, [as ROS1 `rosrun` does](http://wiki.ros.org/rosbash#rosrun). Grounds for this have to be established here and then leveraged in downstream client libraries (e.g. `rclpy` and `rclcpp`).

As a proposal for a syntax and since ROS1 syntax e.g. `_name:=value` would be parsed as a ROS2 remapping rule, we could make use of a dotted `'.'` equivalent `.name:=value` (credits to @sloretz for it). For a usage example, consider:

```sh
ros2 run demo_nodes_cpp talker \
  .bool_param:=true \
  .int_param:=1 \
  .float_param:=1.2 \
  .str_param:='...' \
  .bool_array_param:="[true, false, true]" \
  .int_array_param:="[1, 2, 3]" \
  .float_array_param:="[1.0, 2.0, 3.0]" \
  .str_array_param:="['a', 'b', 'c']" \
  .nested.int_param:=0 \
  .nested:="{str_param: '...'}"
```

which would roughly equate to the following YAML file:

```yaml
bool_param: true
int_param: 1
float_param: 1.2
str_param: '...'
bool_array_param: [true, false, true]
int_array_param: [1, 2, 3]
float_array_param: [1.0, 2.0, 3.0]
str_array_param: ['a', 'b', 'c']
nested:
  int_param: 0
  str_param: '...'
```

#### Implementation considerations

As a likely prerequisite, #246 has to be implemented first.

jhdcs, 2019-06-17:
Is this the same thing as #370?

wjwwood, 2019-06-24:
I would prefer to see a more explicit parameter marker than `.`. I understand that it neatly disambiguates `_foo:=bar` (hidden topic `_foo` remaps to `bar`) from `_foo:=bar` (override parameter `foo` to have a value of `bar`), but if we're going to diverge from ROS 1 anyways, I'd like to see a more explicit flag, as I said before. Maybe something like `-p<key>:=<value>`, at the risk of overlapping with user arguments in someway.

I also do not care for using `.` as I would still like to see parameters go back to using `/` as their separator at some pointer (necessitating node namespaces change to `.` instead of `/` too). I know we already rely on the `.` in existing examples and stuff like that, but I can't help but hesitate before making further dependence on it if it can be avoided.

hidmic, 2019-07-04:
> Is this the same thing as #370?

Yes, it is a duplicate.

> I also do not care for using . as I would still like to see parameters go back to using / as their separator at some pointer (necessitating node namespaces change to . instead of / too).

I understand. Over the course of last week, @wjwwood and I have been discussing about ROS 2 resource addressing in general (be them topics, services, actions, parameters, nodes). As a result of that, I've opened https://github.com/ros2/design/pull/241 and https://github.com/ros2/design/pull/242. 

> Maybe something like -p<key>:=<value>, at the risk of overlapping with user arguments in someway.

Hmm, I'm not a big fan of the `-p` prefix but I'm not really into the operator modification I propose on https://github.com/ros2/design/pull/242 either. Ideas and alternatives are more than welcome.

hidmic, 2020-03-19:
This is complete since:

- [x] https://github.com/ros2/rcl/pull/477
- [x] https://github.com/ros2/rcl/pull/483
- [x] https://github.com/ros2/rcl/pull/491

Closing.

closed 2020-03-19 by hidmic