ros2/rcl #469: Memory leak in test_subscription__rmw_fastrtps_cpp

vmayoral, 2019-07-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- DDS implementation:
  - FastRTPS

#### Actual behavior

When launching the tests compiled with Asan enabled:

<details><summary><b>test_subscription__rmw_fastrtps_cpp</b> result</summary>

```bash
./test_subscription__rmw_fastrtps_cpp
Running main() from /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from TestSubscriptionFixture__rmw_fastrtps_cpp
[ RUN      ] TestSubscriptionFixture__rmw_fastrtps_cpp.test_subscription_nominal
[       OK ] TestSubscriptionFixture__rmw_fastrtps_cpp.test_subscription_nominal (1026 ms)
[ RUN      ] TestSubscriptionFixture__rmw_fastrtps_cpp.test_subscription_nominal_string
[       OK ] TestSubscriptionFixture__rmw_fastrtps_cpp.test_subscription_nominal_string (1014 ms)
[----------] 2 tests from TestSubscriptionFixture__rmw_fastrtps_cpp (2040 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (2040 ms total)
[  PASSED  ] 2 tests.

=================================================================
==12118==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 144 byte(s) in 1 object(s) allocated from:
    #0 0x7f9770f9ab50 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb50)
    #1 0x7f97706e6465 in __default_allocate /opt/ros2_asan_ws/src/ros2/rcutils/src/allocator.c:35
    #2 0x7f9770c7825d in rcl_subscription_init /opt/ros2_asan_ws/src/ros2/rcl/rcl/src/rcl/subscription.c:153
    #3 0x55979db3bde8 in TestSubscriptionFixture__rmw_fastrtps_cpp_test_subscription_nominal_Test::TestBody() /opt/ros2_asan_ws/src/ros2/rcl/rcl/test/rcl/test_subscription.cpp:135
    #4 0x55979dbbd88b in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2447
    #5 0x55979dbaf6b7 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2483
    #6 0x55979db5b559 in testing::Test::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2522
    #7 0x55979db5c984 in testing::TestInfo::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2703
    #8 0x55979db5d528 in testing::TestCase::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2825
    #9 0x55979db78639 in testing::internal::UnitTestImpl::RunAllTests() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:5216
    #10 0x55979dbc033e in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2447
    #11 0x55979dbb1980 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2483
    #12 0x55979db753cd in testing::UnitTest::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:4824
    #13 0x55979db4891c in RUN_ALL_TESTS() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:2370
    #14 0x55979db48862 in main /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/src/gtest_main.cc:36
    #15 0x7f976f4f6b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

Indirect leak of 192 byte(s) in 1 object(s) allocated from:
    #0 0x7f9770f9c7d8 in operator new(unsigned long, std::nothrow_t const&) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe07d8)
    #1 0x7f97709a1b00 in rmw_create_subscription /opt/ros2_asan_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_subscription.cpp:161
    #2 0x7f9770c783b2 in rcl_subscription_init /opt/ros2_asan_ws/src/ros2/rcl/rcl/src/rcl/subscription.c:160
    #3 0x55979db3bde8 in TestSubscriptionFixture__rmw_fastrtps_cpp_test_subscription_nominal_Test::TestBody() /opt/ros2_asan_ws/src/ros2/rcl/rcl/test/rcl/test_subscription.cpp:135
    #4 0x55979dbbd88b in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2447
    #5 0x55979dbaf6b7 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2483
    #6 0x55979db5b559 in testing::Test::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2522
    #7 0x55979db5c984 in testing::TestInfo::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2703
    #8 0x55979db5d528 in testing::TestCase::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2825
    #9 0x55979db78639 in testing::internal::UnitTestImpl::RunAllTests() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:5216
    #10 0x55979dbc033e in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2447
    #11 0x55979dbb1980 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2483
    #12 0x55979db753cd in testing::UnitTest::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:4824
    #13 0x55979db4891c in RUN_ALL_TESTS() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:2370
    #14 0x55979db48862 in main /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/src/gtest_main.cc:36
[... 82 lines cut ...]
    #1 0x7f97706e6465 in __default_allocate /opt/ros2_asan_ws/src/ros2/rcutils/src/allocator.c:35
    #2 0x7f976e4d7e8c in rmw_allocate /opt/ros2_asan_ws/src/ros2/rmw/rmw/src/allocators.c:29
    #3 0x7f97709a1d30 in rmw_create_subscription /opt/ros2_asan_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_subscription.cpp:181
    #4 0x7f9770c783b2 in rcl_subscription_init /opt/ros2_asan_ws/src/ros2/rcl/rcl/src/rcl/subscription.c:160
    #5 0x55979db3bde8 in TestSubscriptionFixture__rmw_fastrtps_cpp_test_subscription_nominal_Test::TestBody() /opt/ros2_asan_ws/src/ros2/rcl/rcl/test/rcl/test_subscription.cpp:135
    #6 0x55979dbbd88b in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2447
    #7 0x55979dbaf6b7 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2483
    #8 0x55979db5b559 in testing::Test::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2522
    #9 0x55979db5c984 in testing::TestInfo::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2703
    #10 0x55979db5d528 in testing::TestCase::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2825
    #11 0x55979db78639 in testing::internal::UnitTestImpl::RunAllTests() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:5216
    #12 0x55979dbc033e in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtes [... cut ...]
    #14 0x55979db753cd in testing::UnitTest::Run() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/./src/gtest.cc:4824
    #15 0x55979db4891c in RUN_ALL_TESTS() /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:2370
    #16 0x55979db48862 in main /opt/ros2_asan_ws/install-asan/gtest_vendor/src/gtest_vendor/src/gtest_main.cc:36
    #17 0x7f976f4f6b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: 515 byte(s) leaked in 7 allocation(s).
```

</details>

#### Expected behavior
Tests running without leaks.

y-okumura-isp, 2020-01-28:
It may be a very very small problem, I'd like to work on it.

In `TEST_F(CLASSNAME(TestSubscriptionFixture, RMW_IMPLEMENTATION), test_subscription_nominal)`,
`rcl_subscription_t subscription` is initialized three times but `rcl_publisher_fini` only last time in this function.

By using local scope as below, the test is passed in my environment.
If OK, I'll send PR.

```cpp
  { // added
    rcl_subscription_t subscription = rcl_get_zero_initialized_subscription();
    ret = rcl_subscription_init(&subscription, this->node_ptr, ts, topic, &subscription_options);
    ASSERT_EQ(RCL_RET_OK, ret) << rcl_get_error_string().str;
    OSRF_TESTING_TOOLS_CPP_SCOPE_EXIT({
      rcl_ret_t ret = rcl_subscription_fini(&subscription, this->node_ptr);
      EXPECT_EQ(RCL_RET_OK, ret) << rcl_get_error_string().str;
    });
    EXPECT_EQ(strcmp(rcl_subscription_get_topic_name(&subscription), expected_topic), 0);
  } // added

  // Test is_valid for zero initialized subscription
  {  // added
    rcl_subscription_t subscription = rcl_get_zero_initialized_subscription(); // val type added
    EXPECT_FALSE(rcl_subscription_is_valid(&subscription));
    rcl_reset_error();
  }  // added
```

y-okumura-isp, 2020-01-30:
BTW asan reports illegal memory usage for other tests such as `test_graph__rmw_fastrtps_dynamic_cpp`.
I'm fixing them (to pass all test in asan is my immediate plan).

y-okumura-isp, 2020-01-31:
Next I send PR for test_arguments.

y-okumura-isp, 2020-02-03:
PR for test_publisher.

y-okumura-isp, 2020-02-03:
PR for test_count_matched

ivanpauno, 2020-02-13:
Fixed by https://github.com/ros2/rcl/commit/19d53bc9e5110f4b66f5f57cce16fd5404daa18b.

closed 2020-02-13 by ivanpauno