ros2/rcl #472: rcl package fails to cross compile

potentialdiffer, 2019-07-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 (VM)
- Installation type:
  - Source
- Version or commit hash:
  - 1bf11ed0c2b53090f990262ba4003d63cb1e5460
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue

Official Cross Compilation Instruction: https://index.ros.org/doc/ros2/Tutorials/Cross-compilation/#cross-compiling-examples-for-arm

#### Expected behavior
Successful cross compiling
#### Actual behavior
The compilation fails when trying to compile rcl package:
```
Starting >>> rcl
Starting >>> rosidl_runtime_py
Finished <<< rosidl_runtime_py [1.05s]                                                               
--- stderr: rcl                                                                                      
CMake Warning:
  Manually-specified variables were not used by the project:

    SECURITY

/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: warning: librosidl_typesupport_fastrtps_c.so, needed by /root/cc_ws/ros2_ws/install/lib/librmw_fastrtps_dynamic_cpp.so, not found (try using -rpath or -rpath-link)
/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: warning: librosidl_typesupport_fastrtps_cpp.so, needed by /root/cc_ws/ros2_ws/install/lib/librmw_fastrtps_dynamic_cpp.so, not found (try using -rpath or -rpath-link)
/root/cc_ws/ros2_ws/install/lib/librmw_fastrtps_dynamic_cpp.so: undefined reference to `rosidl_typesupport_fastrtps_cpp::wstring_to_u16string(std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> > const&, std::__cxx11::basic_string<char16_t, std::char_traits<char16_t>, std::allocator<char16_t> >&)'
/root/cc_ws/ros2_ws/install/lib/librmw_fastrtps_dynamic_cpp.so: undefined reference to `rosidl_typesupport_fastrtps_c::wstring_to_u16string(std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> > const&, rosidl_generator_c__U16String&)'
/root/cc_ws/ros2_ws/install/lib/librmw_fastrtps_dynamic_cpp.so: undefined reference to `rosidl_typesupport_fastrtps_c::u16string_to_wstring(rosidl_generator_c__U16String const&, std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >&)'
/root/cc_ws/ros2_ws/install/lib/librmw_fastrtps_dynamic_cpp.so: undefined reference to `rosidl_typesupport_fastrtps_cpp::u16string_to_wstring(std::__cxx11::basic_string<char16_t, std::char_traits<char16_t>, std::allocator<char16_t> > const&, std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/service_fixture__rmw_fastrtps_dynamic_cpp] Error 1
make[1]: *** [test/CMakeFiles/service_fixture__rmw_fastrtps_dynamic_cpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rcl	[ Exited with code 2 ]
--- stderr: geometry_msgs                                                          
CMake Warning:
  Manually-specified variables were not used by the project:

    SECURITY

---
Aborted  <<< geometry_msgs

Summary: 133 packages finished [12min 51s]
  1 package failed: rcl
  1 package aborted: geometry_msgs
  113 packages had stderr output: action_msgs actionlib_msgs ament_cmake ament_cmake_auto ament_cmake_clang_format ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_flake8 ament_cmake_gmock ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_nose ament_cmake_pclint ament_cmake_pep257 ament_cmake_pep8 ament_cmake_pyflakes ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_uncrustify ament_cmake_xmllint ament_index_cpp ament_lint_auto ament_lint_common builtin_interfaces class_loader composition_interfaces connext_cmake_module console_bridge_vendor example_interfaces fastcdr fastrtps_cmake_module geometry_msgs gmock_vendor gtest_vendor kdl_parser launch_testing_ament_cmake libcurl_vendor libyaml_vendor lifecycle_msgs opensplice_cmake_module orocos_kdl osrf_testing_tools_cpp pendulum_msgs pluginlib poco_vendor python_cmake_module rcl rcl_interfaces rcl_logging_log4cxx rcl_logging_noop rcpputils rcutils resource_retriever rmw rmw_connext_cpp rmw_connext_shared_cpp rmw_implementation rmw_implementation_cmake rmw_opensplice_cpp ros_environment rosgraph_msgs rosidl_adapter rosidl_cmake rosidl_default_generators rosidl_default_runtime rosidl_generator_c rosidl_generator_cpp rosidl_generator_dds_idl rosidl_generator_py rosidl_parser rosidl_typesupport_c rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rosidl_typesupport_cpp rosidl_typesupport_fastrtps_c rosidl_typesupport_fastrtps_cpp rosidl_typesupport_interface rosidl_typesupport_introspection_c rosidl_typesupport_introspection_cpp rosidl_typesupport_opensplice_c rosidl_typesupport_opensplice_cpp rttest shared_queues_vendor sqlite3_vendor std_msgs std_srvs test_interface_files test_launch_testing test_msgs test_osrf_testing_tools_cpp tinydir_vendor tinyxml2_vendor tinyxml_vendor tlsf uncrustify_vendor unique_identifier_msgs urdf urdfdom urdfdom_headers yaml_cpp_vendor
  95 packages not processed
```
#### Additional information
Related issue in cross_compile repo: https://github.com/ros2/cross_compile/issues/17

Disabling test compilation lead in successful compilation, as stated in the other issue, there might be a problem with the cmake files.

nuclearsandwich, 2019-07-25:
> as stated in the other issue, there might be a problem with the cmake files.

Could you please elaborate on the problem you think is present or, ideally, submit a pull request which corrects the problem.

zihui0325, 2019-12-28:
I also have the same problem when I cross compile. It fails when cross compiling `tf2_ros` and `rclcpp_components` package.
The error log
```
[2.013s] aarch64-linux-gnu/bin/ld: warning: librosidl_typesupport_fastrtps_cpp.so, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
[2.014s] aarch64-linux-gnu/bin/ld: warning: librosidl_typesupport_fastrtps_c.so, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
[2.014s] aarch64-linux-gnu/bin/ld: warning: libfastrtps.so.1, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
[2.015s] aarch64-linux-gnu/bin/ld: warning: libssl.so.3, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
[2.015s] aarch64-linux-gnu/bin/ld: warning: libcrypto.so.3, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
[2.015s] aarch64-linux-gnu/bin/ld: warning: libfastcdr.so.1, needed by /home/r18119/ros_hisi/dashing/install/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so, not found (try using -rpath or -rpath-link)
[2.016s] aarch64-linux-gnu/bin/ld: warning: libPocoFoundation.so.64, needed by /home/r18119/ros_hisi/dashing/install/lib/librosidl_typesupport_c.so, not found (try using -rpath or -rpath-link)
```
It seems like that `libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so` and `libbuiltin_interfaces__rosidl_typesupport_fastrtps_cpp.so` don't link to dependencies properly. I check the  ros `install/lib` directory, and all the warning libs are exist.