ros2/rcl #476: Undefined reference to rcl_logging functions

alsora, 2019-07-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source 
- Version or commit hash:
  - master
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```
#include "rclcpp/rclcpp.hpp"
#include "rcl/logging_rosout.h"
#include <rcl/logging.h>
using namespace std::chrono_literals;

class LoggerNode : public rclcpp::Node
{
public:
  LoggerNode()
  : Node("logger_node")
  {
    //auto allocator = rcl_get_default_allocator();
    //rcl_ret_t status = rcl_logging_rosout_init(&allocator);
    rcutils_logging_set_output_handler(rcl_logging_rosout_output_handler);
   
    auto timer_callback =
      [this]() -> void {
        RCLCPP_INFO(this->get_logger(), "Hello World");
      };
    timer_ = this->create_wall_timer(500ms, timer_callback);
  }

private:
  rclcpp::TimerBase::SharedPtr timer_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<LoggerNode>());
  rclcpp::shutdown();
  return 0;
}

```

#### Expected behavior

I create a node with a custom logging output handler (in this case it should only log to `rosout` topic).

#### Actual behavior

The above code doesn't build as I get a linker error:
```
undefined reference to `rcl_logging_rosout_output_handler'
```

This function is marked as `RCL_PUBLIC`, so it should be accessible.

Note that if I comment the following line in `rcl/CMakeLists.txt` the compilation succeeds.
```
rcl_set_symbol_visibility_hidden(${PROJECT_NAME} LANGUAGE "C")
```

#### Additional notes

On the other hand, the function `rcl_logging_rosout_init` is marked as `RCL_LOCAL`, thus it's hidden and I can't use it in my code.
Assuming that the `rcl_logging_rosout_output_handler` has to be accessible somehow, there should also be a way for checking if it has already been initialized and to initialize it.

This issue is linked to 

https://answers.ros.org/question/329532/ros2-how-to-programmatically-set-rosout-logging-handler/
https://discourse.ros.org/t/set-symbol-visibility-to-hidden-for-rmw-and-rcl-packages/7981/5?u=alsora

wjwwood, 2019-08-22:
@alsora can this be closed now that the pr's have been merged? Or is there more to do?

alsora, 2019-08-22:
Yes, the PRs fixes this issue.

closed 2019-08-22 by alsora