ros2/rcl #481: Support parameter descriptions in yaml parser

bpwilcox, 2019-08-07:
----
## Feature request
Add support to parse parameter descriptions in yaml file.

#### Feature description

Given that parameters must now be declared and have optional ParameterDescriptors, it would be an enhancement to support parsing of a parameter description from the yaml file format. This could be convenient for users to not have to specify descriptions in code or to change their parameter settings without recompiling, but another example use case would be to control read-only or range validation for nodes with `NodeOptions.automatically_declare_from_overrides` set as true (such as a "parameter blackboard" ). Since only the default descriptors are currently declared for parameters passed via yaml, any other node can change the value to anything, which may not be desired

This feature would allow users to specify optional descriptions in the yaml file, aligning with the fields in `rcl_interfaces/ParmeterDescriptor.msg`.

An example format could be:
``` yaml
node:
  ros__parameters:
    param1: value
      read_only: true
      description: "this value is read-only"
   param2: value
      from_value: min_value
      to_value: max_value
```

#### Implementation considerations

- Requires matching changes in rclcpp to create ParameterDescriptor objects from parsed yaml and pass to Node parameter overrides

hidmic, 2019-11-21:
@bpwilcox could you further expand on a use case for this feature? A node's implementation usually relies on the existence of some parameters, in which case I think they should explicitly declare them and not rely on an external YAML file to do so -- particularly for client libraries in languages that are statically typed and will likely fail ungracefully if things are not the way they should.

hidmic, 2019-11-21:
Hmm, https://github.com/ros2/rclcpp/issues/807 suggests a parameter blackboard i.e. not an application-specific node but a generic parameter container. Is that what you had in mind?

bpwilcox, 2019-11-21:
@hidmic The parameter blackboard is a use-case that I believe well deserves this feature, but I believe even application-specific nodes could benefit.

>  A node's implementation usually relies on the existence of some parameters, in which case I think they should explicitly declare them and not rely on an external YAML file to do so
>
I believe this would be the case as-is, nodes relying on certain parameters may explicitly declare them, but already the YAML provides an opportunity to overwrite these parameters. As it pertains to parameter descriptors, you can overwrite (or declare if not already provided) key descriptions for paramters as they fit the system. Some example use-cases:
- set certain values to read-only for a particular run of a node
- adapt integer/floating-point ranges to match a variety of different robot/context configurations
  - if using a parameter tuning plugin, this helps define allowable ranges to expose
- provide additional documentation on parameter descriptions/constraints
- fill in any description fields not already declared in source code

If there is any sensitivity for an application to not accept any run-time configurations from the YAML, there is already a flag in `rclcpp::Node::declare_parameter`, `ignore_override`, users can set to reject parameter overwrites:

Another "extended" use-case I'm currently experimenting with is automatic parameter tuning of ROS2 parameters, where I'm aligning the parameter descriptors for ranges to define the bounds of candidate parameters in each optimization test.