ros2/rcl #482: Update API documentation rcl for dashing

JanStaschulat, 2019-08-07:
## Feature request

Changes in rcl API are not documented for Dashing.
#### Feature description
For example, ...

rcl_wait_set_init()
https://github.com/ros2/rcl/blob/master/rcl/include/rcl/wait.h#L131
The number_of_events is added as a parameter, but this parameter is not documented!

rcl_take() 
a new parameter is introduced: rmw_subscription_allocation_t * allocation
Documentation forthis parameter is missing. The unit test is only updated with nullptr argument, which is not helpful.
https://github.com/ros2/rcl/blob/master/rcl/include/rcl/subscription.h#L266
Unit test for rcl_take( only using nullptr for allocation parameter):
https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_subscription.cpp#L180
https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_subscription.cpp#L231

maryaB-osr, 2020-03-09:
closed because dashing is a bit older now, but it's still in use so reopening, also because I need to check if this was updated for eloquent+

nuclearsandwich, 2020-11-20:
I keep slipping on this task since docs are currently built manually. I won't slip this time.

closed 2022-06-21 by clalancette