ros2/rcl #489: rcl_yaml_param_parser MAX_STRING_SIZE is quite small --> navigation crashes

oneattosecond, 2019-08-20:
## Bug report

**Required Info:**

- Operating System:
  - MacOS 10.14.x
- Installation type:
  - source
- Version or commit hash:
  - 0.7.5
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - bt_navigator

#### Steps to reproduce issue
check out the repos at a long path like /Users/myname/coolstuff/morestuff/rcl/

build navigation stack, launch navigation via typical launch script

#### Expected behavior
everything works and no crashes seen

#### Actual behavior
bunch of nodes (lifecycle_manager, bt_navigator, etc) crash with an error in rcl_yaml_param_parser:

[lifecycle_manager-1] libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Failed to parse parameters from file '/var/folders/0m/441s_ghd5mz21g_f0bb597hc0000gn/T/tmpwnxxqstg': Scalar value at line 48 is bigger than 128 bytes, at ~/ros2_osx_dashing/src/ros2/rcl/rcl_yaml_param_parser/src/parser.c:849

snippet of /var/folders/0m/441s_ghd5mz21g_f0bb597hc0000gn/T/tmpwnxxqstg:
bt_navigator:
ros__parameters:
bt_xml_filename: /really_long_path_here/install/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml

#### Additional information
Would it be possible to increase the string limit to 256 or 512? Or possibly find another way to handle the limits? Having nodes crash due to a string limit seems rather sub-optimal. I had to move my repo to a shorter, less convenient path (and recompile various nodes due to wrong symlinks) to get past this

oneattosecond, 2019-08-20:
see https://github.com/ros-planning/navigation2/issues/1044 for some background if needed.

dirk-thomas, 2019-08-20:
Duplicate of the existing PR #487.

dirk-thomas, 2019-08-20:
The existing PR proposes to increase the limit to 256. If you have a use for which this isn't sufficient please comment on the PR with more information to consider increasing the limit even further.

closed 2019-08-20 by dirk-thomas

jdlangs, 2019-09-18:
I'd like to request this issue be reopened. A new maximum size of 256 isn't really changing the core problem of an unnecessarily small parameter size limit. Parameters in ROS1 were often quite huge, such as the entire contents of files (e.g., URDF/SRDF).

Furthermore, does this need to be escalated to a design discussion? I'm pretty sure if a limit of 128/256 bytes had been proposed in the design repo there would have been instant push back and we could have resolved this long ago.

oneattosecond, 2019-09-18:
@jdlangs totally agree, there shouldn't be limits like this. However I do think it's a dupe of #490, and I'm hoping #490 actually just removes arbitrary limits instead of increasing said limits.

schornakj, 2019-10-11:
What was the original motivation behind `MAX_STRING_SIZE`? What would happen if the checks against `MAX_STRING_SIZE` were removed entirely?

Building off @jdlangs 's comment, an example of ROS1 functionality that relies on arbitrary-length string parameters are [the robot description loader launch files in `ur_description`](https://github.com/ipa-led/universal_robot/blob/kinetic-devel/ur_description/launch/ur10_upload.launch). More generally, I think that `xacro`-based URDF generation at runtime essentially requires this capability.

clalancette, 2019-10-11:
For the people using ROS 2 in real-time applications, they'd like to not have memory allocations except at very specific and knowable times (i.e. initialization).  So this fixed-sized string was put in to limit allocations.

I agree that what we currently have is too small.  One compromise we might consider is having fixed-size up to a certain size (like the 256 bytes), and then doing allocations if we see things would get larger than that.  If we did that, I think we'd *also* have to add a flag to control the behavior, since if you are doing real-time you want to be sure you won't do allocations by accident.

jdlangs, 2019-10-11:
@clalancette, your response is quite unsatisfying to me because it suggests you were perfectly willing to sacrifice general usability for a (relatively) narrow use-case. You also mention "compromise" but what we're talking about here is a substantial regression from ROS1. I hope you don't disagree with what I said above, that if it had been suggested in a design repo PR that string parameters should have a max length of 128, there is no way it would have survived the community review.

As an immediate actionable followup, one of us could open a PR that just removes the string limit. Is that likely to be accepted or would there be pushback that the real-time use-case is not accommodated?

dirk-thomas, 2019-10-11:
> one of us could open a PR that just removes the string limit. Is that likely to be accepted or would there be pushback that the real-time use-case is not accommodated?

It will likely not get accepted (if it just removes the limit) since that would break the currently supported real time use case.

clalancette, 2019-10-14:
> @clalancette, your response is quite unsatisfying to me because it suggests you were perfectly willing to sacrifice general usability for a (relatively) narrow use-case. You also mention [...]

I agree, 128 (and 256) is way too short.  I don't think I've argued otherwise.  But real-time *is* something we want to support, so we need to think about solutions that can support real-time but still provide the functionality others need.  Hence my proposal above.  It's clearly not the only proposal, and I'm open to hearing other suggestions on how we could accommodate both of the use cases.

oneattosecond, 2019-10-14:
given the discussion, should we re-open this issue? or at least update https://github.com/ros2/rcl/pull/487 which currently only bumps the size up.

jdlangs, 2019-10-14:
The currently open #244 pretty much covers this; discussion should probably be shifted over there.