ros2/rcl #510: provide option to not create logging publisher

wjwwood, 2019-09-30:
## Feature request

#### Feature description

Currently the node unconditionally creates a publisher for logging:

https://github.com/ros2/rcl/blob/d655147310e3e63aff1555ad40fca1bb9cb7783e/rcl/src/rcl/node.c#L358

This is potentially undesirable in systems which do not need this and have an rmw layer that aggressively preallocates storage for publishers and subscriptions.

There should be some way to control whether or not this occurs when creating the node, or possibly globally.

#### Implementation considerations

We should look for places where we assume this publisher exists for each node and ensure it falls back gracefully when it is disabled.

fujitatomoya, 2019-10-02:
fundamentally it looks like following flag is true,
https://github.com/ros2/rcl/blob/d655147310e3e63aff1555ad40fca1bb9cb7783e/rcl/src/rcl/logging.c#L43

but we do not want to make publisher instance, am i understanding correct?

fujitatomoya, 2019-10-10:
@wjwwood 

could you take a look at https://github.com/ros2/rcl/pull/514 ? when you got time.

wjwwood, 2019-10-17:
> IMO, having the ability of enabling/disabling rosout logging in each node individually is desired.

Definitely, and so I would be fine with leaving this open and/or closing it but opening a new issue.

fujitatomoya, 2019-10-17:
@ivanpauno @wjwwood 

okay, got it.

what about,
- add rosout enable/disable flag in NodeOption as each node setting.
- and we can also keep the global switch g_rcl_logging_rosout_enabled.

or you already have proposal?

ivanpauno, 2019-10-18:
> @ivanpauno @wjwwood
> 
> okay, got it.
> 
> what about,
> 
> * add rosout enable/disable flag in NodeOption as each node setting.
> * and we can also keep the global switch g_rcl_logging_rosout_enabled.
> 
> or you already have proposal?

Sounds like a good plan!

Barry-Xu-2018, 2019-10-24:
@ivanpauno @fujitatomoya @wjwwood 

> what about,
> add rosout enable/disable flag in NodeOption as each node setting.
> and we can also keep the global switch g_rcl_logging_rosout_enabled.

I implemented them.  
Could you review my patches ?  
#532 and ros2/rclcpp#900

fujitatomoya, 2019-10-24:
@Barry-Xu-2018 

requesting fix for https://github.com/ros2/rcl/pull/532#pullrequestreview-306297107

Barry-Xu-2018, 2019-10-24:
@fujitatomoya 

> requesting fix for #532 (review)

Thanks for your review.   
But I have a question on your comments.   
Could you check it https://github.com/ros2/rcl/pull/532#issuecomment-545745768

fujitatomoya, 2019-12-16:
@wjwwood @ivanpauno 

is there anything we could help you via this thread? if not, could you close this issue?

Best,

ivanpauno, 2019-12-16:
> is there anything we could help you via this thread? if not, could you close this issue?

Thanks for the reminder. I think that everything has been done.
Thanks for the contributions @fujitatomoya @Barry-Xu-2018 !

closed 2019-12-16 by ivanpauno