ros2/rcl #526: [YAML parser]: Remove arbitrary limits of MAX_NUM_PARAMS_PER_NODE and MAX_NUM_NODE_ENTRIES

clalancette, 2019-10-21:
In #524, we removed the maximum string limit in the yaml parser.  There are also two other hard-coded limits in there:

* [MAX_NUM_NODE_ENTRIES](https://github.com/ros2/rcl/blob/cdf1b4d17dd9058e19d62373c76c8d1ec70f97c0/rcl_yaml_param_parser/src/parser.c#L72) - This is the maximum number of nodes that can be in a single parameters file, and is currently set to 256
* [MAX_NUM_PARAMS_PER_NODE](https://github.com/ros2/rcl/blob/cdf1b4d17dd9058e19d62373c76c8d1ec70f97c0/rcl_yaml_param_parser/src/parser.c#L73) - This is the maximum number of parameters that a single node can have, and is currently set to 512

We should look at removing both of these limitations as well, as they are kind of arbitrary.  However, removing them is slightly tricky since they are used at initialization time to allocate the entire array of `char *` that can ever be used during parsing.  To remove them, we'll have to implement a scheme where we do reallocations as we are parsing to make room for the new entries.  The really naive, and potentially poorly-performing, way to do this is to reallocate on every new entry.  The slightly smarter way to do it is to reallocate in batches, where when we need to do a reallocation we reallocate 512 of them at a time or something.  This also requires tracking a capacity vs. used, so is a bit more complicated.

wjwwood, 2019-11-21:
Closing since https://github.com/ros2/rcl/pull/524 was merged.

clalancette, 2019-11-21:
This is actually a separate issue than what was fixed in #524.  That PR was about removing the arbitrary string limit, while this one is about opening the *other* arbitrary limits in the parser.  It's lower priority since they are set pretty high, but it is still something we should look at doing.  So I'll reopen this.

wjwwood, 2019-12-12:
Consider documenting on a "limits page", like this one: https://sqlite.org/limits.html

iuhilnehc-ynos, 2020-09-14:
@clalancette 

I'd like to contribute to this feature.

Main idea: I'm going to remove these two MACROS for rcl_node_params_s/rcl_params_s  by adding a capability that will be automatically updated with num_params[num_nodes] * 2 if num is not enough which adding an item.

clalancette, 2020-09-14:
@iuhilnehc-ynos Fantastic, great to hear.  I'll assign this ticket to you; please feel free to request my review on any PR that you open for this.

PR #802 (2020-09-24): Remove MAX_NUM_PARAMS_PER_NODE and MAX_NUM_NODE_ENTRIES limitation.
Fixes #526
closed 2020-09-24 by clalancette