ros2/rcl #527: Perform substitutions when parsing yaml

mogumbo, 2019-10-22:
In ROS1, yaml files could contain substitutions and it was very useful. I'd like to have this feature in ROS2 as well so that params in my yaml files can point to resources in specific packages like this:
```
my_node:
  ros__parameters:
    data_file: $(find my_package)/data/my_data_file.dat
```

wjwwood, 2019-11-21:
I'm going to close this in favor of this approach https://github.com/ros2/launch_ros/issues/96, and we can re-open later if we decide different.

closed 2019-11-21 by wjwwood

aravindSwamy94, 2023-03-04:
Hi, Is there any update in this, How do we add substitutions in yaml file? This feature was really useful in ROS1 
```
wheel_seperation: $(env WHEEL_SEPERATION)