ros2/rcl #528: Parameter grouping doesn't work with new command line syntax

ivanpauno, 2019-10-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from source
- Version or commit hash:
  - master
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclcpp (probably the same happens in rclpy)

#### Steps to reproduce issue

```bash
ros2 run demo_nodes_cpp parameter_blackboard --ros-args -p "some_lists.some_integers:=[1, 2, 3, 4]"
```

#### Expected behavior
The node is created with the specified parameters.

#### Actual behavior
An error:
```
[ERROR] [rcl]: Failed to parse global arguments
```

#### Additional information

Connected with https://github.com/ros2/ros2_documentation/pull/376.

ivanpauno, 2019-10-23:
@hidmic FYI

hidmic, 2019-10-23:
As we spoke offline, this is due to an inconsistency in how the YAML parser in `rcl_yaml_param_parser`  names parameters and how `rcl` expects them: the former uses dots (`.`) to flatten dictionaries, the latter expects slashes (`/`).  IIUC we want to go towards slashes, there's even [a design document](https://github.com/ros2/design/pull/241) on it. But it's likely too late to make it uniform everywhere. 

So, I'll change `rcl` to use dots now and I'll leave a TODO to go back to slashes for F-Turtle. @dirk-thomas @mjcarroll @wjwwood for awareness.

jacobperron, 2019-10-25:
@hidmic Can this be closed since #530 ?

mjcarroll, 2019-11-07:
@hidmic another bump, can this one be closed now?

hidmic, 2019-11-14:
Whoops, completely lost the notification. Yes, we can close.

closed 2019-11-14 by hidmic