ros2/rcl #543: ros2 eloquent debug model

cjsJack, 2019-11-30:
when I use ros2 eloquent debug version, in windows, if I use some command, like ros2 or ros2 topic list, there will be some error, 
 
## Bug report
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'load': No module named 'netifaces'
Failed to load entry point 'standalone': No module named 'netifaces'
Failed to load entry point 'types': No module named 'netifaces'
Failed to load entry point 'unload': No module named 'netifaces'
Failed to load entry point 'start': No module named 'netifaces'
Failed to load entry point 'status': No module named 'netifaces'
Failed to load entry point 'stop': No module named 'netifaces'
Failed to load entry point 'get': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'nodes': No module named 'netifaces'
Failed to load entry point 'set': No module named 'netifaces'
Failed to load entry point 'info': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'delete': No module named 'netifaces'
Failed to load entry point 'describe': No module named 'netifaces'
Failed to load entry point 'dump': No module named 'netifaces'
Failed to load entry point 'get': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'set': No module named 'netifaces'
Failed to load entry point 'create_key': No module named 'cryptography'
Failed to load entry point 'create_keystore': No module named 'cryptography'
Failed to load entry point 'create_permission': No module named 'cryptography'
Failed to load entry point 'distribute_key': No module named 'cryptography'
Failed to load entry point 'generate_artifacts': No module named 'cryptography'
Failed to load entry point 'generate_policy': No module named 'netifaces'
Failed to load entry point 'list_keys': No module named 'cryptography'
Failed to load entry point 'call': No module named 'netifaces'
Failed to load entry point 'find': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'type': No module named 'netifaces'
Failed to load entry point 'bw': No module named 'netifaces'
Failed to load entry point 'delay': No module named 'netifaces'
Failed to load entry point 'echo': No module named 'netifaces'
Failed to load entry point 'find': No module named 'netifaces'
Failed to load entry point 'hz': No module named 'netifaces'
Failed to load entry point 'info': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'pub': No module named 'netifaces'
Failed to load entry point 'type': No module named 'netifaces'
usage: ros2 topic [-h] [--include-hidden-topics]
                  Call `ros2 topic <command> -h` for more detailed usage. ...
ros2 topic: error: argument Call `ros2 topic <command> -h` for more detailed usage.: invalid choice: 'list' (choose from
**Required Info:**

- Operating System:
 Windows 10 
- Installation type:
binaries
- Version or commit hash:
ros2 eloquent
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
 rclcpp

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

dirk-thomas, 2019-12-02:
Please follow the install instructions to install the missing dependency `netifaces`: https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Install-Binary/#install-dependencies

closed 2019-12-02 by dirk-thomas

cjsJack, 2019-12-03:
Thanks for solution, it can't work also, in ros2-eloquent-release version
model can work well, but when I use ros2-eloquent-debug version model, it
can't work, like this
C:\Windows\system32>ros2
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'load': No module named 'netifaces'
Failed to load entry point 'standalone': No module named 'netifaces'
Failed to load entry point 'types': No module named 'netifaces'
Failed to load entry point 'unload': No module named 'netifaces'
Failed to load entry point 'start': No module named 'netifaces'
Failed to load entry point 'status': No module named 'netifaces'
Failed to load entry point 'stop': No module named 'netifaces'
Failed to load entry point 'get': No module named 'netifaces'
Failed to load entry point 'list': No module named 'netifaces'
Failed to load entry point 'nodes': No module named 'netifaces'
Failed to load entry point 'set': No module named 'netifaces'
Failed to load entry point 'info': No module named 'netifaces'

I already install dependency follow tutorial, and the following is
dependency:
choco list  --local-only
Chocolatey v0.10.15
asio 1.12.1
chocolatey 0.10.15
chocolatey-core.extension 1.3.5.1
chocolatey-windowsupdate.extension 1.0.4
cmake 3.15.5
cmake.install 3.15.5
cunit 2.1.3
eigen 3.3.4
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20181019
KB3033929 1.0.5
KB3035131 1.0.3
log4cxx 0.10.0
python 3.7.5
python3 3.7.5
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.23.27820
vcredist2013 12.0.40660.20180427
     thanks, jack

dirk-thomas, 2019-12-03:
Did you install `netifaces` using `pip` and the Python debug interpreter?

cjsJack, 2019-12-04:
Hi:
Thanks again, if I install  netifaces using pip and the Python debug
interpreter  by commanding python_d -m pip install --force-reinstall
netifaces, it also can't work, error will like the following:
C:\Windows\system32>ros2
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'load': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'standalone': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'types': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'unload': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'start': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'status': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'stop': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'get': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'nodes': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'set': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'info': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'delete': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'describe': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'dump': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'get': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'set': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'create_key': No module named
'cryptography.hazmat.bindings._constant_time'
Failed to load entry point 'create_keystore': No module named
'cryptography.hazmat.bindings._constant_time'
Failed to load entry point 'create_permission': No module named
'cryptography.hazmat.bindings._constant_time'
Failed to load entry point 'distribute_key': No module named
'cryptography.hazmat.bindings._constant_time'
Failed to load entry point 'generate_artifacts': No module named
'cryptography.hazmat.bindings._constant_time'
Failed to load entry point 'generate_policy': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'list_keys': No module named
'cryptography.hazmat.bindings._constant_time'
Failed to load entry point 'call': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'find': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'type': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'bw': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'delay': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'echo': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'find': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'hz': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'info': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'list': Module use of python37.dll conflicts
with this version of Python.
Failed to load entry point 'pub': Module use of python37.dll conflicts with
this version of Python.
Failed to load entry point 'type': Module use of python37.dll conflicts
with this version of Python.
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2 is an extensible command-line tool for ROS 2.
optional arguments:
  -h, --help            show this help message and exit
Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
  daemon     Various daemon related sub-commands
  doctor     Check ROS setup and other potential issues
  interface
  launch     Run a launch file
  lifecycle  Various lifecycle related sub-commands
  msg        Various msg related sub-commands
  multicast  Various multicast related sub-commands
  node       Various node related sub-commands
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  srv        Various srv related sub-commands
  test       Run a ROS2 launch test
  topic      Various topic related sub-commands
  trace      Trace ROS nodes to get information on their execution
  wtf        Use `wtf` as alias to `doctor`
  Call `ros2 <command> -h` for more detailed usage.

And also if I run rviz2 in ros2-eloquent-debug-binary, it also can't work,
error like this:

[image: 捕获.PNG]

jacobperron, 2019-12-04:
@dirk-thomas Correct me if I'm mistaken, but if you want to run ROS 2 Python code with the debug interpreter, not all of the dependencies can simply be installed via pip. For our CI jobs, we [install manually created debug wheels](https://github.com/ros2/ci/blob/950d454332432f077b9c9acbed798d65f879791a/ros2_batch_job/__main__.py#L508-L512) for certain packages (including `cryptography` and `netifaces`).
If this is correct, maybe we should add a section to the docs for installing dependencies for Windows debug.

@cjsJack So I think you could try installing our debug wheels for those packages.

cjsJack, 2019-12-04:
Thanks for support, I install the packages following the doc, when I run
ros2, it is ok,
[image: 捕获.PNG]
But when I run ros2 topic list, it will also appear some error, like this:
[image: 捕获.PNG]
And also, when I run rviz2, it still fail, like this:
[image: 捕获.PNG]
Because our team using ros2-eloquent-debug binary in windows to debug some
project, and develop some function, so we must use ros2-eloquent-debug
version, in order to  convenient debug.
Thanks again,
                 jack

jacobperron, 2019-12-04:
@cjsJack We're unable to see the images you are posting. Please copy-paste the output when possible.

cjsJack, 2019-12-04:
Thanks for support, I install the packages following the doc, when I run
ros2, it is ok,
'
https://github.com/ros2/ros2/releases/download/cryptography-archives/cffi-1.12.3-cp37-cp37dm-win_amd64.whl',
# required by cryptography
'
https://github.com/ros2/ros2/releases/download/cryptography-archives/cryptography-2.7-cp37-cp37dm-win_amd64.whl',

'
https://github.com/ros2/ros2/releases/download/lxml-archives/lxml-4.3.2-cp37-cp37dm-win_amd64.whl',

'
https://github.com/ros2/ros2/releases/download/netifaces-archives/netifaces-0.10.9-cp37-cp37dm-win_amd64.whl',

'
https://github.com/ros2/ros2/releases/download/numpy-archives/numpy-1.16.2-cp37-cp37dm-win_amd64.whl
'

But when I run ros2 topic list, it will also appear some error, like this:
Failed to load entry point 'create_key': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'create_keystore': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'create_permission': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'distribute_key': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'generate_artifacts': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'generate_policy': Module use of python37.dll
conflicts with this version of Python.
Failed to load entry point 'list_keys': Module use of python37.dll
conflicts with this version of Python.
C:\dev\ros2-eloquent-debug\Lib\site-packages\ros2cli\node\strategy.py:26:
ResourceWarning: unclosed <socket.socket fd=820,
family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0>
  if is_daemon_running(args):
ResourceWarning: Enable tracemalloc to get the object allocation traceback
/parameter_events
/rosout

And also, when I run rviz2, it still fail, like this:
C:\dev\ros2-eloquent-debug\Package>rviz2
[ERROR] [rviz2]: Ogre::InternalErrorException::InternalErrorException:
Could not load dynamic library
C:\dev\ros2-eloquent-debug\opt\rviz_ogre_vendor\bin\RenderSystem_GL_d.
System Error: 操作成功完成。
 in DynLib::load at
C:\J\workspace\packaging_windows_debug\ws\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreDynLib.cpp
(line 117)

jacobperron, 2019-12-06:
@cjsJack To clarify, did you try running a commands like the following?

```
python_d -m pip install https://github.com/ros2/ros2/releases/download/cryptography-archives/cffi-1.12.3-cp37-cp37dm-win_amd64.whl
python_d -m pip install https://github.com/ros2/ros2/releases/download/cryptography-archives/cryptography-2.7-cp37-cp37dm-win_amd64.whl
python_d -m pip install https://github.com/ros2/ros2/releases/download/lxml-archives/lxml-4.3.2-cp37-cp37dm-win_amd64.whl
python_d -m pip install https://github.com/ros2/ros2/releases/download/netifaces-archives/netifaces-0.10.9-cp37-cp37dm-win_amd64.whl
python_d -m pip install https://github.com/ros2/ros2/releases/download/numpy-archives/numpy-1.16.2-cp37-cp37dm-win_amd64.whl
```

cjsJack, 2019-12-06:
yes, but also will appear some error, when I run ros2, ros2 topic list or
rviz2

jacobperron, 2019-12-06:
Based on the error message "Module use of python37.dll", it sounds like it's possible you might be mixing debug and release packages. But it's difficult to say without being able to reproduce your environment.

liuchangji, 2024-07-30:
Try this, it works for me.
pip install --upgrade --force-reinstall netifaces