ros2/rcl #545: Move filesystem utility to rcutils

BorjaOuterelo, 2019-12-03:
## Feature request

#### Feature description

Move file system dependant functionality to `rcutils` will increase the ROS 2 possible targets.

https://github.com/ros2/rcl/blob/3a5c3a34191bd95491cb72c7458b9409452a28f0/rcl/src/rcl/node.c#L320

In this concrete case, I propose to `move rcl_get_secure_root` API to `rcutils`.
This way, rcl will be "easy" to cross-compile for systems with no filesystem.
Keeping the core mostly platform-independent.

In `rcutils` there are already other utilities for the filesystem.

#### Implementation considerations

We propose to have and `rcutils_get_secure_root` API and be called in the same place as `rcl_get_secure_root` is called right now. These two pull request implements the proposed solution over dashing:

- https://github.com/ros2/rcutils/pull/195
- https://github.com/ros2/rcl/pull/544

jacobperron, 2019-12-03:
It's probably better to consolidate discussion. To reiterate my comment (https://github.com/ros2/rcl/pull/544#issuecomment-561336558):

> I believe there was a past discussion about relocating/refactoring the security directory implementation, but I'm unable to find it at the moment. In any case, I don't think that moving the functionality to rcutils is the right thing to do simply because rcutils is a place for ROS-agnostic helpers for the ROS code base. Instead, perhaps creating a new package to contain the core security functions makes sense (e.g. rcl_security). I'd have to think about it some more.

jacobperron, 2019-12-03:
I found the past discussion I was thinking about in #332. In particular, I agree with this https://github.com/ros2/rcl/pull/332#issuecomment-448072660:

> I think the best solution (long term solution) would be to remove all of the use of filesystem and environment variable usage related to security from rcl itself, instead passing in the functions to do this from the client libraries or by using plugins (similar to how Connext's security works). The reason for this would be to make rcl more flexible, so that on systems which use a different way of configuring security or storing keys can customize their solution with a plugin or in the user code rather than having to modify rcl itself. Also for users who are not using security but are on machines which do not have things like filesystems.

BorjaOuterelo, 2019-12-04:
I do agree with the #332 comment, isolate rcl from security configuration giving the users the right to implement theirs.

> keep rcl "about the client library"

and

> Also for users who are not using security but are on machines which do not have things like filesystems.

are appealing to us.

Is there any roadmap for such changes?

jacobperron, 2019-12-04:
I'm not aware of any explicit roadmap, but we can start by using this ticket for tracking progress.

jacobperron, 2020-02-20:
I've closed the connected `rcutils` PR and re-titled this issue. Rather than moving security utilities to `rcutils`, I think we should move the implementation to a separate package that clients can choose to opt out of. E.g. `rcl_security`