ros2/rcl #550: Linking error on Raspbian Buster

Jesus805, 2019-12-08:
## Bug report

**Required Info:**

- Operating System:
Raspbian Buster Lite
- Installation type:
From source
- Version or commit hash:
ROS2 eloquent
- DDS implementation:

- Client library (if applicable):

#### Steps to reproduce issue
I am building from source using the following ROS2 link.
https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/
However, I generated the repos using
```rosinstall_generator --deps --rosdistro eloquent ros_base > ros2.repos```

#### Expected behavior
To build
#### Actual behavior
```
--- stderr: rcl
/usr/bin/ld: ../librcl.so: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: ../librcl.so: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../librcl.so: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../librcl.so: undefined reference to `__atomic_exchange_8'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/client_fixture__rmw_fastrtps_cpp.dir/build.make:145: test/client_fixture__rmw_fastrtps_cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:838: test/CMakeFiles/client_fixture__rmw_fastrtps_cpp.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
#### Additional information
This issue is similar to issue 171 in [rcutils](https://github.com/ros2/rcutils/issues/171)

dirk-thomas, 2019-12-08:
> This issue is similar to issue 171 in rcutils

This looks more like a duplicate of that ticket. Can you please clarify what you think is different compared to the already existing ticket?

Jesus805, 2019-12-08:
Hm, I didn't know different repositories shared the same tickets. I suppose it is a duplicate and can be closed. I just wanted to report that this issue is also happening in rcl

dirk-thomas, 2019-12-08:
Since this looks like a systematic issue and it probably applies to each and every repo I think a single ticket is sufficient.

closed 2019-12-08 by dirk-thomas

hangcui1201, 2020-01-13:
It seems the fix method for rcutils and rcl are different. One is adding target_link_libraries(test_atomics_executable ${PROJECT_NAME} -latomic) in the CMakeLists.txt for rcutils. How about rcl? Thanks!

Jesus805, 2020-01-13:
@hangcui1201 I was able to build by adding ```add_link_options(-latomic)``` in each CMakeLists.txt that wouldn't build.

For example in `rcl/CMakeLists.txt` line 31
```
if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
  add_link_options(-latomic)
endif()
```

hangcui1201, 2020-01-14:
@Jesus805 Thank you so much!