ros2/rcl #551: `rcl_configure_logging` and `rcl_logging_fini` are called multiple times when creating multiple contexts

ivanpauno, 2019-12-09:
## Feature request

#### Feature description

Currently, if many contexts are created, [rcl_logging_configure](https://github.com/ros2/rcl/blob/024c3ee50d9b79a85665efe1c191cc1225026519/rcl/src/rcl/init.c#L125) is called many times, overriding preexisting configurations silently.
Same happens with [rcl_logging_fini](https://github.com/ros2/rcl/blob/024c3ee50d9b79a85665efe1c191cc1225026519/rcl/src/rcl/init.c#L188). That destroys the logger when the first context is destructed.

The logger should be only initialized once, and initialization of future contexts should be ignored logging a warning (or handling the error somehow).

#### Implementation considerations

To avoid destructing the logger at the first call, an initialization count can be implemented.
The logger should be destroyed after `rcl_logging_destroy` was called the same times that init.

#### Note

Check if something similar may be happening with other "global" configurations.

fujitatomoya, 2019-12-24:
@ivanpauno @wjwwood 

let me take care of this, will get back to you.

Best

fujitatomoya, 2019-12-25:
@ivanpauno 

my thoughts are the followings,

- logging confiration is based on global arguments, so should be initalized and finalized once in the process space.
- rcl_logging_configure will be changed into rcl_logging_init to keep the name consistency for rcl_logging_fini.
- Thread-Safe is not supported, just to get along with current implementation.
- reference counter is provided to keep the consistency, how many times initialized and finalized.
- reference counter and implementation is declared in static scope in logging.

just let me know if i got anything missing and wrong.

Best,

ivanpauno, 2020-01-02:
> let me take care of this, will get back to you.

Sounds good, thanks!

> logging confiration is based on global arguments, so should be initalized and finalized once in the process space.

Yeap, I agree. Alternativly, we could have some context dependent logging configs, but I don't think that's a good idea (and definetly, much more work).

> rcl_logging_configure will be changed into rcl_logging_init to keep the name consistency for rcl_logging_fini.

That sounds reasonable, though it's not required. You can skip that in the first iteration, in order to minimize changes in upstream packages.

> Thread-Safe is not supported, just to get along with current implementation.

I would ensure thread safety in logging init-fini functions, to avoid strange runtime errors.

> reference counter is provided to keep the consistency, how many times initialized and finalized.
> reference counter and implementation is declared in static scope in logging.

Sounds good. Besides that, it would be good to show a warning when logging is initialized more than once, passing different `argv` options. Maybe, just showing a warning when `argv` is not `NULL` and logging was already inited.

ivanpauno, 2020-01-16:
@fujitatomoya are you planning to contribute in this one?

fujitatomoya, 2020-01-21:
@ivanpauno 

yes, i am going to do it, i am positive that i can do that in this week.
if urgent, somebody else takes over.

it was just hard to squeeze time...sorry to be late to get back to you.

fujitatomoya, 2020-01-23:
@ivanpauno 

could you share your thoughts on https://github.com/ros2/rcl/pull/560 ?

i am not so sure that we need to ensure thread safe on this function, maybe warning should be enough.

fujitatomoya, 2020-05-19:
@ivanpauno 

i guess that we can close this since https://github.com/ros2/rcl/pull/560 is merged.

ivanpauno, 2020-05-19:
Yes, https://github.com/ros2/rcl/pull/579 https://github.com/ros2/rclpy/pull/518 and https://github.com/ros2/rclcpp/pull/998 fixed this.

It would nice too modify the logger so that its configuration isn't global, but that's a different issue.

closed 2020-05-19 by ivanpauno