ros2/rcl #552: No documentation for clock types

nickolai-voyage, 2019-12-19:
## Feature request

Please add documentation for the different clock types, i.e. RCL_STEADY_TIME, RCL_SYSTEM_TIME

#### Feature description
There's no documentation about the differences between these times. I'm guessing that STEADY_TIME is monotonically increasing just based on the name, but it would be really nice to have actual documentation. Something that gets into the doxygen docs would be great.

jacobperron, 2020-01-03:
Here's the design document describing the clock types: https://design.ros2.org/articles/clock_and_time.html

@nickolai-voyage Would you like to contribute a pull request adding API documentation?

fujitatomoya, 2020-01-23:
@nickolai-voyage 

is there any reason you picked up RCL_STEADY_TIME, RCL_SYSTEM_TIME in particular?

i guess that https://design.ros2.org/articles/clock_and_time.html should be okay for application since rcl is not exactly the front-end. what do you think?

PR #578 (2020-03-02): Document clock types
Resolves #552
commit ca645250 (2020-03-02): Document clock types (#578)

Resolves #552
closed 2020-03-02 by jacobperron