ros2/rcl #553: SIGABRT parsing yaml config file (double free detected)

guru-florida, 2019-12-22:
## Bug report
rclcpp::init(...) causing SIGABRT while parsing config yaml that is well formed yaml but not valid ROS2 parameters format. For example, if yaml has only a single underscore in ros__parameters.

- Operating System: Ubuntu 19.04 on x64 and also reproduced on RPI4 using Ubuntu 19
- Installation type: source
- Version or commit hash: eloquent branch
- DDS implementation: default
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
remove an underscore from ros__parameters. Or have a yaml without node-name, such as just:
`joints: 17`

#### Expected behavior
Should produce an error saying the yaml file is malformed, or at least not crash but node would not receive parameters.

#### Actual behavior
Program produces the following output:
```
free(): double free detected in tcache 2
Aborted (core dumped)

```

#### Additional information
Using `--cmake-args -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS="-fsanitize=address"` produces this output:
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==24262==ERROR: AddressSanitizer: SEGV on unknown address 0x00002dfffff1 (pc 0x7f3437322962 bp 0x00002dfffff1 sp 0x7ffecf4d24c0 T0)
==24262==The signal is caused by a WRITE memory access.
    #0 0x7f3437322961  (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x28961)
    #1 0x7f3437406014 in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10c014)
    #2 0x7f3436d0a6de in __default_deallocate (/opt/ros2/eloquent/install/rcutils/lib/librcutils.so+0x46de)
    #3 0x7f34367308e8 in rcl_yaml_node_struct_fini (/opt/ros2/eloquent/install/rcl_yaml_param_parser/lib/librcl_yaml_param_parser.so+0x28e8)
    #4 0x7f343676ab8e in rcl_arguments_fini (/opt/ros2/eloquent/install/rcl/lib/librcl.so+0xfb8e)
    #5 0x7f3436769ab0 in rcl_parse_arguments (/opt/ros2/eloquent/install/rcl/lib/librcl.so+0xeab0)
    #6 0x7f3436772f19 in rcl_init (/opt/ros2/eloquent/install/rcl/lib/librcl.so+0x17f19)
    #7 0x7f34370f5d69 in rclcpp::Context::init(int, char const* const*, rclcpp::InitOptions const&) (/opt/ros2/eloquent/install/rclcpp/lib/librclcpp.so+0x3d9d69)
    #8 0x7f3437212fa7 in rclcpp::init(int, char const* const*, rclcpp::InitOptions const&) (/opt/ros2/eloquent/install/rclcpp/lib/librclcpp.so+0x4f6fa7)
    #9 0x55c93cebce1a in main /home/guru/src/humanoid/ros2/humanoid/src/lss_joint_publisher/src/lss_joint_states.cpp:96
    #10 0x7f3436943b6a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x26b6a)
    #11 0x55c93ceba619 in _start (/home/guru/src/humanoid/ros2/humanoid/build/lss_joint_publisher/lss_joint_states+0xf619)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x28961)

jacobperron, 2020-01-03:
If a parameter file is malformed, it might actually be safer to fail instead of letting rcl_init continue. Passing incorrect values to a node (e.g. defaults) could be bad for an application, resulting in unexpected behavior. 

We should still fix the crash and instead return an appropriate error code.

guru-florida, 2020-01-03:
Yes, I agree with SIGABRT preferably with a stderr line preceding it would be the safe bet, or std::exception with what() filled in.

jacobperron, 2020-01-16:
I tried to reproduce the issue, and the double free only occurs when using the deprecated syntax e.g. 

        ros2 run demo_nodes_cpp talker __params:=bogus_config.yaml

If we use the recommended syntax then I don't see the double free:

        ros2 run demo_nodes_cpp talker --ros-args --params-file bogus_config.yaml

And we get a more informative error message:

    'Couldn't parse params file: '--params-file bogus_config.yaml'. Error: Cannot have a value before ros__parameters at line 7 ...

I also noticed that the double free does not occur in Dashing. The bug must have been introduced when the `__params` syntax was deprecated in Eloquent. Since the deprecated syntax has been removed since Eloquent, this bug is specific to Eloquent.

guru-florida, 2020-01-17:
Thanks @jacobperron for the fix! Regarding the deprecated syntax, this bug surfaced using python launch file description...perhaps that is using the old syntax and should be updated to use the new?

jacobperron, 2020-01-17:
@guru-florida Are you able to share a launch file that I can use to reproduce the issue?

jacobperron, 2020-01-17:
Nevermind, I guess this is the fix for the issue you were seeing: https://github.com/ros2/launch_ros/pull/106

It will be backported to Eloquent and should be fixed in the next patch release:  https://github.com/ros2/launch_ros/pull/115

jacobperron, 2020-01-29:
This was fixed by https://github.com/ros2/rcl/pull/556.

PR #556 (2020-01-28): Fix behavior when provided a malformed parameters file
Fixes #553. 

* Don't finalize parameters struct in rcl_parse_yaml_file function
The function assumes that the structure has already been initialized, so it should be the callers
responsibility to finalize it. Otherwise, this may lead to a double free as reported in #553.
* Return an error code if there's a malformed parameters file
This restores behavior that was lost when the '__params:=' syntax was deprecated in #495.
* An extra guard was also added when finalizing the parameters struct.
closed 2020-01-29 by jacobperron