ros2/rcl #555: yaml parser raises an error "Sequence should be of same type" when meets .nan in list of doubles

iquater, 2020-01-16:
ros2 launch raises an exception while parsing list with .nan value

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - dashing
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp, rclpy

#### Steps to reproduce issue
Create python or cpp package:
```python
import rclpy
from rclpy.node import Node

class TestNode(Node):
    def __init__(self):
        super().__init__("parameter_holder")

def main(args=None):
    rclpy.init(args=args)
    parameter_holder = TestNode()
    rclpy.spin(parameter_holder)
    parameter_holder.destroy_node()
    rclpy.shutdown()
    return 0
```
launch file:
```python
from launch.launch_description import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(package='reprod',
             node_executable='param_holder',
             node_name='parameter_holder',
             output='screen',
             parameters=[
                 "/tmp/test.yaml"
             ])
    ])
```
configuration file
```yaml
parameter_holder:
    ros__parameters:
        test_list: [1.0, 2.0]
        test_list2: [1.0, .nan]
```

#### Expected behavior
Yaml parser should treat .nan as valid double (or float) value.
#### Actual behavior
Yaml parser treats .nan as string and throws exception:

`RuntimeError: Failed to parse yaml params file '/tmp/test.yaml': Sequence should be of same type. Value type 'string' do not belong at line_num 4`

dirk-thomas, 2020-01-16:
Loosely related to ros2/rosidl#351.

ivanpauno, 2020-01-30:
@iquater Thanks for opening the issue. This is not currently supported, but it's a desired feature.
A PR adding support to special floating point values is more than welcome.

closed 2020-09-02 by brawner