ros2/rcl #563: [YAML parser] Support for anchors/aliases

wieset, 2020-01-31:
## Feature request

#### Feature description
Anchors/aliases are useful in YAML to avoid repetition, when several configuration blocks share the same group of parameters.

Example:

``` yaml
test_config: &test
  parameter_1: 123
  parameter_2: 456

device_a:
  name: "a"
  config: *test

device_b:
  name: "b"
  config: *test
```

An anchor (denoted by a leading "&")  precedes either a scalar, a sequence or a mapping. An alias (denoted by a leading "*") corresponds to an anchor of the same name. The parser should insert the anchored block at the position of the alias.

#### Implementation considerations
In libyaml, an ALIAS event is produced when reading an alias. The alias needs to be matched to a previously defined anchor of the same name. Anchors must be checked for existence and uniqueness.

corot, 2025-06-02:
What a shock.... This is available on ROS 1. How tricky would be to port the parsing code?  I have 0 experience modifying the rcl, but if easy I can give it a try

andreas-loeffler, 2026-02-20:
@corot any updates on this?