ros2/rcl #574: Add tests checking when a graph guard condition is triggered

ivanpauno, 2020-02-11:
## Feature request

#### Feature description

There's currently nothing testing when a graph guard condition is triggered, tests should be added.

#### Implementation considerations

Current graph tests could be refactored to first wait a graph condition is triggered, that will also avoid some flakiness of the current tests.

Barry-Xu-2018, 2020-08-27:
@ivanpauno 

> Feature description
>
> There's currently nothing testing when a graph guard condition is triggered, tests should be added.

I want to make clear the requirement of this issue.  

After checking test in [rcl/test/rcl/test_graph.cpp](https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_graph.cpp), existing test cases are to wait graph guard condition triggering after graph is changed (such as calling rcl_publisher_init()).  
e.g.
test case `test_graph_guard_condition_topics` 

https://github.com/ros2/rcl/blob/2daa408ea46bb2a4305a6221d73ea400e4c02d62/rcl/test/rcl/test_graph.cpp#L1271-L1287 

Calling rcl_publisher_init(), rcl_subscription_init(), rcl_subscription_fini() and rcl_publisher_fini() will affect graph.  That is, `graph guard condition` will be triggered. 

While a graph guard condition is triggered, what test do you want to be added ? e.g. check if publisher is added or deleted ?  

> Implementation considerations
>
> Current graph tests could be refactored to first wait a graph condition is triggered, that will also avoid some flakiness of the current tests.

Current test implementation like as below (such as check_graph_state() ) 
```
loop for try 
    Check status (such as the number of publishers, subscribers)
    If not match expected value 
         add graph guard condition to wait_set
         rcl_wait() //Wait for graph changing 
```

Do you mean rcl_wait() should be done (after graph condition trigger) before check status ?

ivanpauno, 2020-09-23:
> test case test_graph_guard_condition_topics

Yeah, that test isn't really reliable (those timeouts may be a problem), but that's the basic idea.
Instead of using a thread, I would call functions that have "graph side-effects" and then wait (with a reasonable timeout) to see if the graph guard condition was triggered, sequentially (one at a time).

> While a graph guard condition is triggered, what test do you want to be added ? e.g. check if publisher is added or deleted ?

Anything that has a "graph side-effect" and isn't currently tested is worth adding.

With a "graph side-effect", I mean anything that can change the output of all the functions defined [here](https://github.com/ros2/rcl/blob/master/rcl/include/rcl/graph.h) and [publisher_count](https://github.com/ros2/rcl/blob/8bcada7b884693644d7ba2e3bd2c22fd9541eba7/rcl/include/rcl/subscription.h#L568)/[subscription_count](https://github.com/ros2/rcl/blob/8bcada7b884693644d7ba2e3bd2c22fd9541eba7/rcl/include/rcl/publisher.h#L603) functions.

> Do you mean rcl_wait() should be done (after graph condition trigger) before check status ?

I don't get the question.

Barry-Xu-2018, 2020-09-25:
@ivanpauno 

Got it.  Thanks for your explanation.   
I or @Ada-King will continue to handle this issue. 

>> Do you mean rcl_wait() should be done (after graph condition trigger) before check status ?
>
>I don't get the question.

You can ignore this unclear question.

Barry-Xu-2018, 2020-09-28:
@ivanpauno 

Please review #811

fujitatomoya, 2020-12-22:
@ivanpauno @Barry-Xu-2018 

can we close this?

closed 2020-12-23 by ivanpauno