ros2/rcl #581: Expose QoS settings for /rosout publisher

de-vri-es, 2020-02-26:
I have noticed that sometimes during node startup, some messages don't arrive at `/rosout` subscribers. I could never replicate this on my own Linux hardware, but I have witnessed it on a Windows PC. Log messages from the first 30 to 100ms would not arrive at a subscriber.

I imagine this could be fixed by changing the QoS settings of the publisher. However, I don't believe we have access to the publisher or it's QoS settings at the moment. If it is true that changing the QoS settings could prevent loosing messages during startup (within reason), could these settings be exposed?

Log messages during node startup can be especially interesting, so it's not nice to loose them.

dirk-thomas, 2020-02-26:
While exposing the QoS settings would be great I think the default still needs to be different. Currently it uses `rmw_qos_profile_default` but something like `rmw_qos_profile_parameters` which changes the depth to 1000 and the duratbility to transient local would be more appropriate.

See #582 for the proposal. @de-vri-es It would be great if you could give this patch a try and comment if that resolves your issue.

de-vri-es, 2020-02-26:
That does sound like a good idea. I'm not very familiar with compiling on Windows, which is where I witnessed the issue, but I'll try my best to test it.

ivanpauno, 2020-03-04:
Deleted the `bug` label after https://github.com/ros2/rcl/pull/582 and #587, the defaults qos are now nicer.
I will left this open, as exposing the qos would still be an improvement.

de-vri-es, 2020-03-04:
Sounds good. Thanks all for the quick action  :)

fujitatomoya, 2020-07-14:
@dirk-thomas @ivanpauno @de-vri-es 
CC: @Barry-Xu-2018 

anybody working on this? if not, we could take care of this thread.

ivanpauno, 2020-07-17:
> anybody working on this? if not, we could take care of this thread.

AFAIK, nobody is working on this.

Usually, people assign themselves when they are working on something (though they might forget :joy:).
We don't usually address issues marked as `help wanted`. Of course, PRs targeting a `help wanted` issue are more than welcomed.

Ada-King, 2020-07-21:
Hi, @ivanpauno @fujitatomoya  I made a PR for this issue, now only rcl/rmw layer was implemented. As for the rclcpp layer, i have already finished a design draft, which we can discuss later, For now, can you take a quick look for [253](https://github.com/ros2/rmw/pull/253) && [722](https://github.com/ros2/rcl/pull/722) in your spare time?

ivanpauno, 2020-08-04:
Solved in https://github.com/ros2/rcl/pull/722

closed 2020-08-04 by ivanpauno