ros2/rcl #586: Create an inner publisher(rosout) in low level API may destroy ROS2's Perfection

reed-lau, 2020-03-03:
https://github.com/ros2/rcl/blob/ca64525029991809ee22dd4f082d6539c41c7f88/rcl/src/rcl/logging_rosout.c#L36

From the crystal version, the node will create an inner publisher with the topic name `/rosout` as a component of the ROS log-system. 
when there are few nodes, it worked ok. But for a relatively complex system. the inner publisher will cost much and connected together under some rmw-implenentation.
In my point of view, creating such a thing in such lower-level API isn't very suitable, although, it can be disabled. Contrarily, if we really need one, we can easily establish it with the final user-level API.

I wish such basic things in ROS2 (eg. node/publisher/subscriber/client/service) could try their best to behave as its concept and with function as simple as possible. Just like the Unix Philosophy: ```Make each program do one thing well ```. I wanna say `make node/publisher/subscriber/client/service do one thing well`.

reed-lau, 2020-03-10:
@dirk-thomas

ivanpauno, 2020-03-10:
@reed-lau I think that doing the rosout publisher opt-in instead of opt-out would be reasonable.
Can you propose a PR doing that?

clalancette, 2020-03-10:
> @reed-lau I think that doing the rosout publisher opt-in instead of opt-out would be reasonable.

My opinion is that *most* users want rosout, and the ones that don't are the exception.  Thus, I think that opt-out is more reasonable.  But others can chime in as well.

ivanpauno, 2020-03-10:
> My opinion is that most users want rosout, and the ones that don't are the exception. Thus, I think that opt-out is more reasonable. But others can chime in as well.

In general I prefer defaulting to the option that creates minimal overhead and then opt-in everything else, than trying to guess what *most* users want (i.e.: following the "zero cost abstraction" principle). But I understand it's a matter of preference.

In the future, we could add something like "ROS profiles", where all this stuff could be configured from an external file (e.g.: specified with an env variable). Then we could have different recommended profiles, like "beginner friendly", "minimal overhead", etc.

reed-lau, 2020-03-10:
it should aim at an industrial level, not a toy. when we use ROS, we need a framework, not a total solution.

clalancette, 2020-03-10:
> it should aim at an industrial level, not a toy.

But we have users who are just getting started in robotics, academics, industrial users, automotive users, government users, etc.  We want to serve all of those groups as well as possible, so my opinion is that we should do the things that make it easy to get started, while still giving the power to do more advanced things for more advanced users.  Hence my opinion that `rosout` be opt-out.

dirk-thomas, 2020-03-10:
> it should aim at an industrial level, not a toy. when we use ROS, we need a decoupled framework, not a total solution.

It is configurable to make it suitable for all kinds of applications. Especially for industrial applications you shouldn't have a problem configuring it to your needs. Why is opt-out a problem for your use case?

> In fact, there're various well-designed logger system, but not `rosout`.

And again, that is why the API is designed in a way which let's you configure your own backend. If you think any part needs improvement please consider to contribute valuable feedback or even pull requests towards it.

ivanpauno, 2020-03-10:
> it should aim at an industrial level, not a toy.

Doing this option opt-in or opt-out wouldn't affect the suitability to be industrial level, as it can be configured properly.

reed-lau, 2020-03-10:
> Doing this option opt-in or opt-out wouldn't affect the suitability to be industrial level, as it can be configured properly.

ok, I'll try my best to learn about it, thank you for answering.

closed 2020-03-19 by claireyywang