ros2/rcl #602: Progress toward Quality Level 1

Blast545, 2020-03-19:
This issue tracks the progression of packages in `rcl` that have been deemed necessary for Quality Level 1 and a 1.0 version level. It follows the outline described in [REP 2004](https://github.com/ros-infrastructure/rep/blob/rep-2004/rep-2004.rst). 

The packages slated for Quality Level 1 for ROS 2 Foxy are:
- rcl
- rcl_action
- rcl_lifecycle
- rcl_yaml_param_parser

# Progress common to all packages:
* [x] Version Policy
    - Follows ROS Core Quality Declaration
* [ ] Version >= 1.0.0
* [x] Change Control Process
    - Follows ROS Core

Documentation
* [x] Declared set of licenses
* [x] Copyright statement in each source file
    - Checked with ament_lint_common

Platform Support
* [x] Supports all tier 1 platforms

# Progress of specific packages
## rcl

Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
* [ ] Quality Declaration document
  - * [First draft](https://github.com/ros2/rcl/pull/566) to be reviewed

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - rcl_interfaces, rcl_logging_spdlog, rcl_yaml_param_parser, rcutils, rmw_implementation, rosidl_generator_c, tracetools
* [ ] Non-ROS dependencies are equivalent level 1
  - libyaml through rcl_yaml_param_parser, others to be removed

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
  - Tests included, requires describing main pkg features. Doxyfile included.
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - API and features to be defined, .hpp headers?
* [ ] Code Coverage:
  - * [x] Must have code coverage tracking for the package. [76%](https://ci.ros2.org/job/ci_linux_coverage/28/cobertura/src_ros2_rcl_rcl_src_rcl/). 
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled
* Extra notes: rcl is going through some important changes removing dependencies and adding features

## rcl_action
Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
* [ ] Quality Declaration document

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - action_msgs, rcl, rcutils, rmw, rosidl_generator_c
* [ ] Non-ROS dependencies are equivalent level 1
  - acquires the same from its ros dependencies

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
  - Tests included, requires describing main pkg features. Doxyfile included.
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - API and features to be defined, .hpp headers?
* [ ] Code Coverage:
  - * [x] Must have code coverage tracking for the package. [77%](https://ci.ros2.org/job/ci_linux_coverage/34/cobertura/src_ros2_rcl_rcl_action_src_rcl_action/). 
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled

## rcl_lifecycle
Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
* [ ] Quality Declaration document

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - lifecycle_msgs, rcl, rcutils, rmw, rosidl_generator_c
* [ ] Non-ROS dependencies are equivalent level 1
  - acquires the same from its ros dependencies

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
  - Tests included, requires describing main pkg features.
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - API and features to be defined, .hpp headers?
* [ ] Code Coverage:
  - * [x] Must have code coverage tracking for the package. [53%](https://ci.ros2.org/job/ci_linux_coverage/34/cobertura/src_ros2_rcl_rcl_lifecycle_src/). 
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled

## rcl_yaml_param_parser
Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
* [ ] Quality Declaration document

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - libyaml_vendor, rcutils
* [ ] Non-ROS dependencies are equivalent level 1
  - libyaml, started discussion for this [here](https://docs.google.com/document/d/18LiYAPO5_d7xuAfwpC7YLmbk8mO29aQf5iIC2hYL2XY/edit).

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
  - Tests included, requires describing main pkg features.
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - API and features to be defined, .hpp headers?
* [ ] Code Coverage:
  - * [x] Must have code coverage tracking for the package. [77%](https://ci.ros2.org/job/ci_linux_coverage/34/cobertura/src_ros2_rcl_rcl_yaml_param_parser_src/. 
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled

jacobperron, 2020-05-12:
FYI, we now have a 1.0.0 release: https://github.com/ros/rosdistro/pull/24843

chapulina, 2020-11-25:
All packages reached level 1 :tada: https://github.com/ros2/rcl/pull/866/

closed 2020-11-25 by chapulina