ros2/rcl #628: Convert sleep_for into count_subscriptions logic

mjcarroll, 2020-04-21:
## Feature request

#### Feature description

Converting from TODO for visibility: https://github.com/ros2/rcl/blob/94b5a1d7d0899aa84a1026e21488bee95e67bbd8/rcl/test/rcl/test_subscription.cpp#L178-L180

There are a few places in the `test_subscription` code (and possibly other places) where we make use of a fixed delay to wait for publications/subscriptions to become available.  This should be converted to make use of `count_publications`/`count_subscriptions` when possible.

Barry-Xu-2018, 2020-04-22:
@mjcarroll 

I have an interest in implementing it.  
I start to implement it.

mjcarroll, 2020-04-22:
Thanks @Barry-Xu-2018!  Looking forward to your pull request, let me know if you run into any issues.

Barry-Xu-2018, 2020-04-23:
@mjcarroll 

I have made PR.  
Please review #631.

jacobperron, 2020-05-15:
Looks like this was resolved in #631.

closed 2020-05-15 by jacobperron