ros2/rcl #713: Call call rcutils_reset_error() even if rcl_lifecycle_transition_init() return RCL_RET_OK in lifecycle test

Barry-Xu-2018, 2020-07-13:
## Bug report

**Required Info:**

- Operating System:
    All
- Installation type:
  Binaries and From source
- Version or commit hash:
   master branch (0de31d04830)
- DDS implementation:
   Any

#### Steps to reproduce issue

Check codes 

https://github.com/ros2/rcl/blob/944068f3d99d6a4b1130fd62e38c69b96f0e759e/rcl_lifecycle/test/test_rcl_lifecycle.cpp#L134-L153

#### Expected behavior

- At line 137 and 142, if rcl_lifecycle_transition_init() return RCL_RET_OK, rcutils_reset_error() should not be called.

- Once rcl_lifecycle_transition_init() is called successfully, rcl_lifecycle_transition_fini() should be called for recycling resource.

#### Actual behavior

- At line 137 and 142, even if rcl_lifecycle_transition_init() return RCL_RET_OK, rcutils_reset_error() is also called.

- rcl_lifecycle_transition_fini() isn't called after rcl_lifecycle_transition_init() is called successfully.

Karsten1987, 2020-07-13:
Thanks for reporting this issue. I think this goes in hand with what we see at the nightly builds as well:
```
05:32:22 1: [ RUN      ] TestDefaultStateMachine.default_init
05:32:22 1: 
05:32:22 1: >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
05:32:22 1: This error state is being overwritten:
05:32:22 1: 
05:32:22 1:   'invalid allocator, at /home/jenkins-agent/workspace/nightly_linux_release/ws/src/ros2/rcl/rcl_lifecycle/src/transition_map.c:91'
05:32:22 1: 
05:32:22 1: with this new error message:
05:32:22 1: 
05:32:22 1:   'can't free transition map, no allocator given
05:32:22 1: , at /home/jenkins-agent/workspace/nightly_linux_release/ws/src/ros2/rcl/rcl_lifecycle/src/transition_map.c:57'
05:32:22 1: 
05:32:22 1: rcutils_reset_error() should be called after error handling to avoid this.
05:32:22 1: <<<
05:32:22 1: [       OK ] TestDefaultStateMachine.default_init (1 ms)
```

do you mind opening a pull request for this?

fujitatomoya, 2020-07-13:
@Barry-Xu-2018 

could you change the title? there are double `call` in the head of the title.

Barry-Xu-2018, 2020-07-14:
@Karsten1987 

> do you mind opening a pull request for this?

Yes. I will fixing it.  
Please assign this issue to me.

Barry-Xu-2018, 2020-07-14:
@fujitatomoya 

> could you change the title? there are double call in the head of the title.

OK. I worry about the title is too long. So I just describe one problem in title.

Barry-Xu-2018, 2020-07-14:
@Karsten1987 

> do you mind opening a pull request for this?
>
> Above log doesn't affect test.
> This isn't a bug. Do you want to not show these error message ?

I find the cause. 
It is bug in test codes. I will fix it.

Barry-Xu-2018, 2020-07-15:
@Karsten1987 

I have updated above comments to explain the problem reported by you (https://github.com/ros2/rcl/issues/713#issuecomment-657746623).  
Please take a look.

fujitatomoya, 2020-07-16:
@Barry-Xu-2018 @Karsten1987 

> https://github.com/ros2/rcl/issues/713#issuecomment-657746623

Since the overwrite condition is notified, I do not see this is a problem but status.

> fail_error_message, fini_error

I think failure of registering transitions and memory leak is different, current code cannot support failure w/o memory leak case. And to support the memory leak case, `rcl_copy_error_string` backed with __rcutils_copy_string would be introduced?

Barry-Xu-2018, 2020-07-16:
@fujitatomoya 

> Since the overwrite condition is notified, I do not see this is a problem but status.

Yes. It should be warning on error state to be overwritten.

> I think failure of registering transitions and memory leak is different, current code cannot support failure w/o memory leak case. And to support the memory leak case, rcl_copy_error_string backed with __rcutils_copy_string would be introduced?

About  `memory leak` mentioned by you,  https://github.com/ros2/rcl/issues/713#issuecomment-657746623 doesn't lead memory leak.   
```c
       fini_error = rcl_get_error_string().str; 
       rcl_reset_error();
     }

       RCL_SET_ERROR_MSG_WITH_FORMAT_STRING(
      "Freeing transition map failed while handling a previous error. Leaking memory!"
      "\nOriginal error:\n\t%s\nError encountered in rcl_lifecycle_transition_map_fini():\n\t%s\n",
      fail_error_message, fini_error);
```
After rclreset_error(), fini_error point to empty string. Error message has been reset.  
This leads `RCL_SET_ERROR_MSG_WITH_FORMAT_STRING` output empty error message.

fujitatomoya, 2020-07-16:
> About memory leak mentioned by you, #713 (comment) doesn't lead memory leak.

i wasn't clear on this, my bad. i do understand what you mean. i was mentioning the "Freeing transition map failed while handling a previous error. Leaking memory!" message.

Barry-Xu-2018, 2020-07-16:
> i wasn't clear on this, my bad. i do understand what you mean. i was mentioning the "Freeing transition map failed while handling a previous error. Leaking memory!" message.

Got it. Even if this error said memory leak, actually failure is because `allocator` is set to nullptr for test.  
transition map isn't initialized successfully since `allocator` is nullptr. So there is no memory leak.

Barry-Xu-2018, 2020-07-16:
@Karsten1987 

About https://github.com/ros2/rcl/issues/713#issuecomment-657746623, the solution mentioned at  https://github.com/ros2/rcl/issues/713#issuecomment-658043362 has problem.  

Below is updated solution.  
Please check it.  
https://github.com/Barry-Xu-2018/rcl/commit/a8e848736b0df5e794fdf43c561da927463cf6e5

Barry-Xu-2018, 2020-07-23:
@Karsten1987 

Since fixing (https://github.com/ros2/rcl/pull/715) was merged, I close this issue.

closed 2020-07-23 by Barry-Xu-2018