ros2/rcl #749: Subscription tests segfaulting in CentOS

nuclearsandwich, 2020-08-15:
## Bug report

**Required Info:**

- Operating System:
  - CentOS Linux
- Installation type:
  - source
- Version or commit hash:
  - master, as of c749682
- DDS implementation:
  - rmw_fastrtps_cpp, rmw_fastrtps_dynamic_cpp, rmw_cyclonedds_cpp

#### Steps to reproduce issue

Run test_subscription tests on CentOS 7.
See [nightly CentOS Linux release](https://ci.ros2.org/view/nightly/job/nightly_linux-centos_release/503/), [debug](https://ci.ros2.org/view/nightly/job/nightly_linux-centos_debug/502/), or [extra rmw](https://ci.ros2.org/view/nightly/job/nightly_linux-centos_extra_rmw_release/503/) for examples.

```
-- run_test.py: extra environment variables:
 - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
-- run_test.py: extra environment variables to append:
 - LD_LIBRARY_PATH+=/home/jenkins-agent/workspace/nightly_linux-centos_release/ws/build/rcl
-- run_test.py: invoking following command in '/home/jenkins-agent/workspace/nightly_linux-centos_release/ws/build/rcl/test':
 - /home/jenkins-agent/workspace/nightly_linux-centos_release/ws/build/rcl/test/test_subscription__rmw_cyclonedds_cpp --gtest_output=xml:/home/jenkins-agent/workspace/nightly_linux-centos_release/ws/build/rcl/test_results/rcl/test_subscription__rmw_cyclonedds_cpp.gtest.xml
Running main() from /home/jenkins-agent/workspace/nightly_linux-centos_release/ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc
[==========] Running 15 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 10 tests from TestSubscriptionFixture__rmw_cyclonedds_cpp
[ RUN      ] TestSubscriptionFixture__rmw_cyclonedds_cpp.test_subscription_init_fini_and_is_valid
[       OK ] TestSubscriptionFixture__rmw_cyclonedds_cpp.test_subscription_init_fini_and_is_valid (4 ms)
[ RUN      ] TestSubscriptionFixture__rmw_cyclonedds_cpp.test_subscription_bad_init
-- run_test.py: return code -11
-- run_test.py: generate result file '/home/jenkins-agent/workspace/nightly_linux-centos_release/ws/build/rcl/test_results/rcl/test_subscription__rmw_cyclonedds_cpp.gtest.xml' with failed test
```

hidmic, 2020-08-18:
I'll take a look.

hidmic, 2020-08-18:
So, the problem is that CentOS deals with function trampolines differently. I haven't dug enough, but it'd almost seem like somehow it keeps a single `.got.plt` section for all shared objects so it we taking the address of a function, it can pass around the address of the function trampoline w/o breaking ODR. Unfortunately, that means we cannot call the original function from within the mock in CentOS.

closed 2020-08-18 by hidmic