ros2/rcl #787: test_rcl_lifecycle failing with only cyclonedds

ivanpauno, 2020-09-03:
Example failure: http://build.ros2.org/view/Rci/job/Rci__nightly-cyclonedds_ubuntu_focal_amd64/

I think this error output is the problem:

> Not all nodes were finished before finishing the context

ivanpauno, 2020-09-03:
@brawner this seem to have been introduced in https://github.com/ros2/rcl/pull/731, is that the case?

ivanpauno, 2020-09-03:
FYI @azeey as buildfarmer

brawner, 2020-09-03:
Thanks for bringing this to my attention. #731 likely introduced the test that uncovered this bug. We've been seeing this happen with cyclone and connext, where additional bugs surface after the fault injection PRs are merged. I'll take a look at it and figure out a bug fix

brawner, 2020-09-03:
> additional bugs surface after the fault injection PRs are merged

That was pure hubris. This test failure was all my doing and there were no additional bugs to fix in rmw_cyclonedds_cpp. It makes me wonder why calling `rcl_shutdown` without calling `rcl_node_fini` worked with the other rmws.

ivanpauno, 2020-09-03:
> It makes me wonder why calling rcl_shutdown without calling rcl_node_fini worked with the other rmws

I think that calling destruction functions in the correct order is UB currently (to give a different example, destroying a publisher after destroying the "parent" node).
In some rmw implementations, doing it in the opposite order might be ok, but in some others it might fail.

PR #788 (2020-09-03): Fix test_rcl_lifecycle
This test had a few issues after digging through it when investigating #787. The root cause of #787 was that the node was not being shutdown during the maybe_fail test, but there was also a few other issues with finalization that led to memory leaks. Lastly to match the style in other tests, I reversed `EXPECT/ASSERT` so that the expected value was on the left-hand side.

If the last style fix is undesired, I can switch that back.

Fixes #787.
closed 2020-09-03 by brawner